arduino/Arduino #3455: Editor shortcut/feature regression.

Chris--A, 2015-07-03:
The introduction of the new editor has removed a handy feature.

See: https://forum.arduino.cc/?topic=333659

I can confirm this working in 1.5.7, but not in a build of the latest source.

ffissore, 2015-07-03:
Yep, didn't know that was possible. It seems like we won't be able to enable it back http://fifesoft.com/forum/viewtopic.php?f=10&t=354&p=826&hilit=column+selection&sid=f53085ef1924edcd746c260cd41f4041#p818

per1234, 2017-07-01:
I'm going to close this as a duplicate of https://github.com/arduino/Arduino/issues/6425. Even though this issue came first, it's not clear what the issue is actually about without looking at the Arduino forum thread. The fifesoft forum gives me a 404 error and I seem to remember it has for some time now.

closed 2017-07-01 by per1234