arduino/Arduino #3456: No such file or directory – #include <file.h>  just a question (1.6.6.)

VaclavSal, 2015-07-03:
This is just a question why, being posted here because I am tired of Arduino.cc forum morons personal insults, beside they are not qualified to intelligently  answer my question anyway. .

But if you have a problem with me posting my question here just skip this post. 

I have x.ino  file with #include “CMain.h” where I  have int main () function processing my application code. It all works just fine, no problem.

When I add #include<address.h> as first line in my x.ino my application compiles and still works.

When I move the #include<address.h> as first line in my CMain.h the compiler cannot find the file  anymore, so I get “No such file or directory”. 

What , why or who is stopping the Windows OS / compiler (?)  search for the <file>? 

Just a question , I got no problem putting all my includes in x.ino file. 

Thanks for your time 
Vaclav

ffissore, 2015-07-03:
I didn't catch that: are you moving `#include "CMain.h"` into your `CMain.h` file?

VaclavSal, 2015-07-03:
See the PS , I cannot put the file  name in less than / greater than sings / brackets. 
I am moving / adding  the #include address.h to the CMain.c file and it cannot find it from there. .

ffissore, 2015-07-03:
Ah ok I see. Use a backtick like ```include <address.h>```. See https://guides.github.com/features/mastering-markdown/

ffissore, 2015-07-03:
Ok, so the answer is that the IDE does not parse includes from non .ino files. You have to include address.h in one of your .ino for the IDE to look for the file and add that library path to the gcc command line (`-I` param)

closed 2015-07-03 by ffissore

VaclavSal, 2015-07-04:
So why does it work with search in parent director  ( #include "header,h" ) files AKA #include "header.h" includes in local files.

#include "CMain.h"  in *inc works,
#include "header.h" in CMain works etc etc. 

I did not mention that search for includes in parent directory ( *.ino) works because it works, hence it is non issue to be publish. 

You have to include address.h in one of your .ino for the IDE to look for the file and add that library path to the gcc command line (-I param)

This is not correct, I did not add any additional path to gcc to make the <> file included.

Isn't that a function of OS ??

" and add that library path to the gcc command line (-I param)"

matthijskooijman, 2015-07-05:
What happens is that the IDE searches all your .ino files for #include lines. For each included header file it found, it tries to find a corresponding library that provides that header file. Then, while compiling, those libraries are added in the include path using `-I`, and the libraries itself are included in the link too.

When you put your include in the .h file, the IDE will not see it and thus not include the needed library in the include path and link.

I hope this clarifies your question.

VaclavSal, 2015-07-05:
No it does not clarify anything. 
Why is IDE bypassing OS "enviroment"  for <> files and processes "" files included in local headers?
Let's just drop this discussion, we are getting nowhere with it and it has a solution. 
The sad part is - IDE chokes and these "solutions" are / will be keeping the applications in "blink without delay"  realm forever.

PaulStoffregen, 2015-07-05:
Ultimately, this is the long-term solution.  Well, that plus the other stuff discussed, which I intended to do as a second phase.

https://github.com/arduino/Arduino/pull/2792

Reworking this for the latest IDE code (which has changed dramatically since March) is on my TO-DO list.  But I don't expect to work on it within the next couple months.  Other (paid) work is currently taking priority over my (unpaid) open source contributions.

Eventually I'll get back to that work.  Until then, you'll have to learn to live with things as they are, or contribute yourself, or get others to do so, or just wait.