arduino/Arduino #3461: --verify for 'release' version of arduino IDE executable on Windows

pfeerick, 2015-07-04:
So, after reading the recent [Adafruilt writeup on Travis CI integration](https://learn.adafruit.com/continuous-integration-arduino-and-you/overview) (and getting it to work for both a library and a sketch github repo, yay!), I thought it would be handy to be able to have  automated sketch verification on Windows also. So I tried the below cl and it didn't work how I expected... I get that of course the 'release' build of the Arduino IDE doesn't output any messages to the console after starting as it has already returned control to the OS... so my question is more along the lines of 'is running the 'debug' exe the officially supported method of accessing --verify and --upload on the Windows platform"? As if it ain't, and you should be able to do this with the release exe, then we've got ourselves a bug!

**Release** (arduino.exe)
``` 
D:\Open Source Programming\arduino-hourly>arduino --board arduino:avr:uno --verify "examples\01.Basics\Blink\Blink.ino"

[splash screen, loading configuration, initialising packages, preparing boards, then nothing]
```

**Debug** (arduino_debug.exe)
```
D:\Open Source Programming\arduino-hourly>arduino_debug --board arduino:avr:uno--verify "examples\01.Basics\Blink\Blink.ino"
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
```

pfeerick, 2015-07-05:
Yes, my mistake... sorry about that @matthijskooijman . I must have submitted #3461 by mistake as I had this one open for a while, and then had another thought about the Travis CI lockup I has having problems with, and then must have submitted this instead of closing it. On the Travis CI linux VM, it locks up at 'preparing boards' if you don't have the sketch in a correctly named sub-folder :( So moving the files into a subfolder fixed that issue. All that remains is the focus of this issue (-- verify / --upload on windows).

ffissore, 2015-07-06:
Using `arduino.exe` for CLI is useless. It will produce a sketch but it won't print any output. This is because of how apps work on windows: they can be either `console` or `gui`. `gui` apps produce no text output
Stick to `arduino_debug` when using CLI

closed 2015-07-06 by ffissore

carlosperate, 2015-07-06:
Out of curiosity, when did this change happened? was it in 1.6.2?

matthijskooijman, 2015-07-06:
#1970 (which is about the splash screen showing, but still related) suggests that arduino_debug.exe has been available since 1.5.7. I'm not sure what the situation before that was - probably just the arduino.exe showing no output.

carlosperate, 2015-07-06:
I'm pretty sure some of the 1.5.x arduino.exe versions would still print their output to the terminal, only difference being the splash screen being shown (which I personally liked, but I can understand the need to run cli commands without gui).