arduino/Arduino #3462: Can not start IDE by double clicking on .ino after update to nightly-windows_2015-07-03

bgma, 2015-07-04:
Hello,
 After updating the IDE to the Arduino-1.6.6-nightly-windows_2015-07-03 from Arduino-1.6.6-nightly-windows_2015-06-20 can not I start by double clicking on .ino / .pde longer.
 I run on W7 pro 64-bit. sp1.
 The problem disappeared when I backed off to 2015-06-20  and came back when I updated to 2015-07-03 again.
See previous: #3331
 /Sincerely Bengt-Göran

ffissore, 2015-07-06:
Isn't this explained in https://github.com/arduino/Arduino/issues/3331#issuecomment-111956379 ?

bgma, 2015-07-06:
Yes it is explained in #3331, but the problem is back again.

ffissore, 2015-07-06:
What problem? There is no installer of 1.6.6 so it is expected that double clicking a .ino file will not open it

ffissore, 2015-07-06:
However, installing 1.6.5 with the installer will set up windows registry, hence double clicking a .ino file will open 1.6.5

bgma, 2015-07-06:
I've now tested!

Arduino-1.6.6-nightly-windows_2015-06-20.zip:
I have uninstalled the Arduino from "Program and Functions"
Deleted in the register
  .ino
  .pde
  arduino

Extract all files to H:\arduino-nightly
Started H:\arduino-nightly\arduino.exe
Double-clicking a .ino file and then displays "open with" dialog
	pointed out H:\arduino-nightly\arduino.exe
The file was opened with ArduinoIDE
Finished ArduinoIDE
Double-clicking a .ino file and the file opens in the ArduinoIDE
DONE 2015-06-20
-----------------------------------------------------------------------------------------------------
Arduino-1.6.6-nightly-windows_2015-07-06.zip:
I have verified that the Arduino is not in the "Program and Functions"
Deleted H:\arduino-nightly
Deleted in the register
  .ino
  (.pde) not found
  arduino

Extract all files to H:\arduino-nightly
Started H:\arduino-nightly\arduino.exe
Double-clicking a .ino file and then displays "open with" dialog
	pointed out H:\arduino-nightly\arduino.exe
The file was not opened
	splash screen appears briefly then nothing more.
Double-clicking a .ino file and the file is not opened
When I double-click the file and check in taskmgr starts jawav.exe * 32 then nothing more.
DONE 2015-07-06
-----------------------------------------------------------------------------------------------------

ffissore, 2015-07-06:
This is expected

ffissore, 2015-07-06:
If you want double clicking a .ino to work, use the installer

bgma, 2015-07-06:
This has worked earlier then 1.6.5-r2 also in 1.6.6 (see above).
Why take away a good feature? In addition, because there is no installer for the nightly-windows.

ffissore, 2015-07-06:
Because it created more harm than good, messing up windows registry. Nightly are for early adopters, to whom we are thankful because they accept a slightly harder user experience than those that wait for releases

bgma, 2015-07-06:
Windows auto-file are handled by windows.
This means that you can specify a .cmd file that do something, or that arduno.exe starts when you double-click a .ino, .pde or .c.
What I have concluded is that you seem to have removed the possibility of sending a filename as parameter. although this has worked in the past.

ffissore, 2015-07-07:
@bgma I'm really sorry, this issue is starting looking like a flame and I don't want to act like a flamer. However, your conclusion is wrong. Indeed you can call arduino.exe PATH TO INO and it will open that sketch. I've removed windows registry manipulation, [see here](https://github.com/arduino/Arduino/commit/f6546aea4527bb0723fac6472c1e10328ebe9473#diff-c2beb43483fc657f9dd1907703e83c93L68) and this is not coming back any time soon.
Want to double click? Use the installer
Closing as invalid

closed 2015-07-07 by ffissore

bgma, 2015-07-07:
Hey,
No hard feelings ...
I think we have not really understood each other.
The problem is solved: the path to arduno.exe need to be in the% PATH%.
Come on it after your last post: PATH TO ...
Missed probably because it was not needed before.

 / Sincerely BG