arduino/Arduino #3463: board.*.tool in boards.txt and tools directives in platforms.txt are ignored

SpenceKonde, 2015-07-04:
In a custom core, I have tried changing the platform.txt definitions for avrdude. No change I make in platform.txt to the tools.* directives (including deleting them entirely) has any effect on behavior.

Similarly, in boards.txt, no matter what I put after 
boardname.upload.tool=
it still stubbornly goes for avrdude.

However, if I omit the line entirely, it refuses to work, even through whatever I pass it is being ignored. 

I've been having a hell of a time with this, and I'd really like to be able to distribute a core with it's own avrdude.conf file, so I can support new chips that are not in the avrdude.conf that comes with arduino. It sounds like this should be possible, but it just doesn't seem to work. Am I doing something wrong? 

boards.txt (relevant excerpt)

```

menu.chip=Chip
menu.clock=Clock
menu.bod=B.O.D.
attinyx41.name=ATtinyx41 
attinyx41.upload.tool=totallynot
attinyx41.bootloader.tool=totallynot
attinyx41.bootloader.unlock_bits=0xFF
attinyx41.bootloader.lock_bits=0xFF
attinyx41.build.core=tiny
attinyx41.menu.chip.841=ATtiny841
attinyx41.menu.chip.841.build.mcu=attiny841
attinyx41.menu.chip.841.upload.maximum_size=8192
attinyx41.menu.chip.841.upload.maximum_data_size=512
attinyx41.menu.chip.441=ATtiny441
attinyx41.menu.chip.441.build.mcu=attiny441
attinyx41.menu.chip.441.upload.maximum_size=4096
attinyx41.menu.chip.441.upload.maximum_data_size=256
attinyx41.menu.chip.241=ATtiny241
attinyx41.menu.chip.241.build.mcu=attiny241
attinyx41.menu.chip.241.upload.maximum_size=2048
attinyx41.menu.chip.241.upload.maximum_data_size=256
attinyx41.menu.clock.16external=16 MHz (external)
attinyx41.menu.clock.16external.bootloader.low_fuses=0xEE
attinyx41.menu.clock.16external.build.f_cpu=16000000L
attinyx41.menu.clock.16external.bootloader.file=empty/empty_all.hex
attinyx41.menu.clock.12external=12 MHz (external)
attinyx41.menu.clock.12external.bootloader.low_fuses=0xEE
attinyx41.menu.clock.12external.build.f_cpu=12000000L
attinyx41.menu.clock.12external.bootloader.file=empty/empty_all.hex
attinyx41.menu.clock.8internal=8 MHz (internal)
attinyx41.menu.clock.8internal.bootloader.low_fuses=0xE2
attinyx41.menu.clock.8internal.build.f_cpu=8000000L
attinyx41.menu.clock.8internal.bootloader.file=empty/empty_all.hex
attinyx41.menu.clock.8external=8 MHz (external)
attinyx41.menu.clock.8external.bootloader.low_fuses=0xEE
attinyx41.menu.clock.8external.build.f_cpu=8000000L
attinyx41.menu.clock.8external.bootloader.file=empty/empty_all.hex
attinyx41.menu.clock.1internal=1 MHz (internal)
attinyx41.menu.clock.1internal.bootloader.low_fuses=0x62
attinyx41.menu.clock.1internal.build.f_cpu=1000000L
attinyx41.menu.clock.1internal.bootloader.file=empty/empty_all.hex
attinyx41.menu.clock.20external=20 MHz (external - overclocked)
attinyx41.menu.clock.20external.bootloader.low_fuses=0xEE
attinyx41.menu.clock.20external.build.f_cpu=20000000L
attinyx41.menu.clock.20external.bootloader.file=empty/empty_all.hex
attinyx41.menu.bod.disable=B.O.D. Disabled
attinyx41.menu.bod.disable.bootloader.extended_fuses=0xFF
attinyx41.menu.bod.disable.bootloader.high_fuses=0xD7
attinyx41.menu.bod.1v8=B.O.D. Enabled (1.8v)
attinyx41.menu.bod.1v8.bootloader.extended_fuses=0xED
attinyx41.menu.bod.1v8.bootloader.high_fuses=0xD7
attinyx41.menu.bod.2v7=B.O.D. Enabled (2.7v)
attinyx41.menu.bod.2v7.bootloader.extended_fuses=0xED
attinyx41.menu.bod.2v7.bootloader.high_fuses=0xD5
attinyx41.menu.bod.4v3=B.O.D. Enabled (4.3v)
attinyx41.menu.bod.4v3.bootloader.extended_fuses=0xED
[... 108 lines cut ...]
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'

```

Yet I still get

```

Sketch uses 382 bytes (9%) of program storage space. Maximum is 4,096 bytes.
Global variables use 9 bytes (3%) of dynamic memory, leaving 247 bytes for local variables. Maximum is 256 bytes.
C:\arduino-1.6.3\hardware\tools\avr/bin/avrdude -CC:\arduino-1.6.3\hardware\tools\avr/etc/avrdude.conf -v -pattiny441 -cusbasp -Pusb -Uflash:w:C:\Users\a2\AppData\Local\Temp\build5639649148410942104.tmp/sketch_jul04a.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 14:56:06
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\arduino-1.6.3\hardware\tools\avr/etc/avrdude.conf"

avrdude: AVR Part "attiny441" not found.

```

(note that attiny441 is defined in the avrdude.conf I'm pointing it at - but that's not the one it's using!)

Thanks

ffissore, 2015-11-20:
@SpenceKonde latest 1.6.6 bundles a new tool for compiling, named `arduino-builder`. Can you give 1.6.6 (or, better, the [hourly](http://www.arduino.cc/en/Main/Software#hourly)) a try and report if it works?

sandeepmistry, 2016-04-13:
I'm closing this due to lack of feedback. Please re-open with requested info if you are still interested.

closed 2016-04-13 by sandeepmistry