arduino/Arduino #3466: Automatically suggesting deleting unused variables.

Avamander, 2015-07-05:
If a user deletes a piece of code that contains a certain variable that is used in no other place why not offer automatic deletion of unused variables before compilation?

VaclavSal, 2015-07-05:
Not such a great idea.
Too much automation during development when blocks of code can be simply commented out and NOT  fully deleted keeps  me aware of what code  I did try and discarded.
I agree prevention is better, but when deleting variables became necessary, and not just wanted for convenience of compiler speed (?) , it may be time to revise your code anyway. 

However, since some other development tools actually warn about unused variables implementing this suggestion may not be that difficult.

carlosperate, 2015-07-05:
If a feature like this is ever implemented I strongly believe it should be an "editor warning" only, so highlight unused variables with no automated deletion.

Avamander, 2015-07-05:
_suggesting deleting unused variables_

carlosperate, 2015-07-05:
> _...why not offer automatic deletion of unused variables before compilation_

Avamander, 2015-07-05:
_offer_

It is up to the user.

Chris--A, 2015-07-06:
If you enable warnings in the preferences, then you get:

> sketch_jul06a.ino:3:5: warning: unused parameter 'x' [-Wunused-parameter]

So it does already suggest that its not needed. 

Like Vaclav, I think It would be a huge mistake to try and get the IDE to parse C++. We are all well aware of the problems encountered with the current IDE pre-processor and its failure to address every possible use case.

For instance, removing the unused variable below will cause the code to become invalid and break.

```arduino
int foo(char x){return 1;}

void setup() {
  foo(2);
}

void loop() {}
```

This problem will also be found when crossing object boundaries as when a library .cpp is compiled it has no idea how a sketch may use it (no idea if a function is called let alone passed any optional parameters).

And sometimes an unused variable is the intention. I wrote these functions below for a meta-programming lib once, and the unused variables are **required**, they are what gives the code any meaning at all:

```arduino
template< typename T, typename U > 
  bool compareTypes( const T &t, const U &u ) throw() 
    { return false; }

template< typename T > 
  bool compareTypes( const T &t, const T &u ) throw() 
    { return true; }
```

ffissore, 2015-07-06:
Since raising warnings level offer such advice to users, closing as worksforme

closed 2015-07-06 by ffissore