arduino/Arduino #3467: Error message about running out of memory is unhelpful.

juj, 2015-07-05:
I'm building a piece of source code, and it is giving me the following build error:

	/avr/bin/ld: C:\Users\clb\AppData\Local\Temp\build4551615865478074668.tmp/MyTest.cpp.elf section `.text' will not fit in region `text'
	/avr/bin/ld: region `text' overflowed by 1910 bytes
	collect2: error: ld returned 1 exit status
	Error compiling.

but there is no indication of what caused the overflow. Instead of giving an unhelpful error message, it would be better to give a dump of used memory in the 'text' region, along with file and line information of where the symbols come from, sorted in descending order of size taken, so that it is possible to analyze what takes up memory and to get ideas where to optimize.

matthijskooijman, 2015-07-06:
Hm, the error message is usually more helpful than this. What IDE version are you using? Could you share the full code?

cmaglie, 2015-07-06:
Dumping all the symbols seems a bit awful to me, I've never seen any IDE with such a feature.

juj, 2015-07-06:
Dumping the symbols sounds much better than expecting the user to play the random guessing game. If dumping the symbols automatically sounds bad because nobody did it before, then I am sure there would be a way to add some other method, e.g. a "Sketch -> Memory Usage Statistics..." to profile it, plus when the error message occurs, point the user to open that menu item.

The issue occurs when trying to use the u8glib library with a LD7032 60x32 OLED display. I installed the library, and uncommented the line

    U8GLIB_LD7032_60x32 u8g(13, 11, 10, 9, 8);	// SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9, RST = 8

from their hello world library example, and built it. I haven't done other changes there. The code is here: https://code.google.com/p/u8glib/source/browse/sys/arduino/HelloWorld/HelloWorld.pde . My Arduino version is 1.6.3.