arduino/Arduino #3468: 1.6.5 fails Gatekeeper check on OS X

don, 2015-07-06:
Arduino 1.6.5 fails to open with Gatekeeper warning

OS X 10.10.4

Arduino 1.6.4 and 1.6.3 download on 15-July-2015 have the same broken behavior. (I though these worked in the past.) https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

Arduino 1.6.2 download on 15-July-2015 works OK.

spctl is happy with the file. No output means it passes.

	$spctl --assess Arduino.app
	
For reference something like Tor fails this check.

	$ spctl --assess /Applications/TorBrowser.app
	/Applications/TorBrowser.app: rejected

codesign thinks the app is signed
	
	$ codesign -dvvv Arduino.app
	Executable=/Users/don/Downloads/Arduino.app/Contents/MacOS/Arduino
	Identifier=cc.arduino.Arduino
	Format=bundle with Mach-O universal (i386 x86_64)
	CodeDirectory v=20200 size=322 flags=0x0(none) hashes=9+3 location=embedded
	Hash type=sha1 size=20
	CDHash=66ea5496d86ed9d9715abb363d5b23612eddeb2d
	Signature size=8505
	Authority=Developer ID Application: ARDUINO SA
	Authority=Developer ID Certification Authority
	Authority=Apple Root CA
	Timestamp=Jun 15, 2015, 5:40:58 AM
	Info.plist entries=24
	TeamIdentifier=7KT7ZWMCJT
	Sealed Resources version=2 rules=12 files=3337
	Internal requirements count=1 size=180
	
opening the file gets a gatekeeper warning

[image]

Here's the log output from Console.app when opening the app and getting the Gatekeeper warning

	7/5/15 10:28:34.290 PM CoreServicesUIAgent[1649]: File /Users/don/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin failed on loadCmd /Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib
	7/5/15 10:28:34.290 PM CoreServicesUIAgent[1649]: Fails dylib check
	7/5/15 10:28:38.053 PM com.apple.xpc.launchd[1]: (com.apple.xpc.launchd.oneshot.0x1000001e.Arduino[1704]) Service exited due to signal: Killed: 9
	7/5/15 10:28:38.054 PM CoreServicesUIAgent[1649]: unexpected message <OS_xpc_error: <error: 0x7fff7b493c60> { count = 1, contents =
		"XPCErrorDescription" => <string: 0x7fff7b493f70> { length = 18, contents = "Connection invalid" }
	}>

Maybe a path got hardcoded to a dylib on the jenkins server? `/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib`

don, 2015-07-06:
Some additional info. I have 2nd Mac running OS X 10.10.4 where Arduino IDE 1.6.5 works fine with Gatekeeper. I installed 1.6.5 on the 2nd machine sometime last week.

I verified there is no gatekeeper exclusion using `spctl --list`

I downloaded latest 1.6.5 from arduino.cc onto this Mac and the new download fails to open with Gatekeeper.

I copied the working Arduino IDE 1.6.5 from the second mac to the first mac and it runs fine. Gatekeeper is happy.

ffissore, 2015-07-06:
It looks more like an error with your gatekeeer rather than an issue with the IDE. MacOSX releases of the IDE are indeed signed

ffissore, 2015-07-06:
/cc @cmaglie

cmaglie, 2015-07-06:
I'm wondering how gatekeeper could match `/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib` instead of the `Arduino.app/Contents/Java/hardware/tools/avr/lib/libusb-1.0.0.dylib`.

@don can you explain it in some way?

don, 2015-07-06:
@ffissore I've tested on 2 machines running 10.10.4 with the same results. I'll test on more machines today. The tools verify that it's signed by gatekeeper is not happy. 

@cmaglie no idea why gatekeeper is looking for that lib in ~jenkins. Probably some obscure setting in xcode.

Can you guys duplicate on any of your machines? The latest 1.6.5 fails Gatekeeper but 1.6.2 works fine?

don, 2015-07-09:
Gatekeeper fails when I download Arduino IDE 1.6.5 on OS X 10.10.4 using Chrome or Safari and click the zip file to unzip.

Gatekeeps works when download Arduino IDE 1.6.5 on OS X 10.9.4.

Gatekeeper works when download Arduino IDE 1.6.5 on OS X 10.9.4 and scp the zip file to a Mac running 10.10.4 and click the zip file to unzip.

The shasum of all the files is the same 
396ab35fd5306dea1760696e11e96f25eaaedd47  arduino-1.6.5-macosx.zip

The "broken" files show the downloaded from http://downloads.arduino.cc when viewing info with CMD + I

If I scp the "broken" zip to myself, the downloaded metadata is removed, and Gatekeeper is happy.

    $ scp  arduino-1.6.5-macosx.zip localhost:/tmp
    $ cd /tmp
    $ open  arduino-1.6.5-macosx.zip
    $ open Arduino.app

So it appears something is wrong with the meta data on 10.10.4. Maybe it's a bug in 10.10.4? Maybe it's something that can be worked around sending downloading to the browser?

cmaglie, 2015-07-15:
@don 
I can finally reproduce this one, I'll check the signing procedure to discover what's happening.
Thanks for the detailed report!

cmaglie, 2015-07-20:
Indeed the problem is in `avrdude_bin` that contains this path `/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/lib/libusb-1.0.0.dylib` hardcoded in some way.

It seems that OSX 10.10.4 has tightened the checks on packaged contents.

Do you know a way to let gatekeeper skip this check?

cmaglie, 2015-07-20:
@don 
I've added a workaround, may you check again with this release file?

http://downloads.arduino.cc/arduino-1.6.5-r3-macosx.zip

closed 2015-07-20 by cmaglie

cmaglie, 2015-07-20:
Positive feedback received.
Feel free to reopen if the issue is still present.

don, 2015-09-17:
@cmaglie just seeing this notification now... http://downloads.arduino.cc/arduino-1.6.5-r3-macosx.zip looks good. Thanks for fixing this!

commit d6c5344b (2015-07-20): Workaround for MacOSX gatekeeper (OSX 10.10.4)

It seems that gatekeeper now applies tighter checks on
libraries dinamically linked from executables included
in a package.

This is a workaround until a proper solution is found.

Fixes #3468