arduino/Arduino #3473: Arduino Logo appears blurry in the toolbar

moritzeto, 2015-07-06:
Hi, 
I've notticed that in my Windows 8.1 computer, the Arduino Logo that appears in the toolbar is very blurry, but in the desktop its all OK. The first image is from the toolbar, and the second one from the desktop.
[image]
[image]

carlosperate, 2015-07-06:
Yeah, it seems like it is using the 16x16 version of the icon. Could it have something to do with the fact that is the same icon resolution used in the top left corner of the window?

ffissore, 2015-07-07:
Yes, indeed. It's ugly. That icon is in file `IDE\application.ico`. AFAIK, it's correct. Nonetheless it's ugly so there is something I don't know. I'm no windows hacker and I need some help

ffissore, 2015-07-07:
Right path is `IDE\lib\arduino_icon.ico`

cmaglie, 2015-07-07:
Maybe it's sufficient to just replace the icon with an hi-res one?

ffissore, 2015-07-07:
That's the conundrum: that `.ico` already includes different resolutions of the same icon. Windows is picking up the wrong one when putting it into the app bar. I don't know why

carlosperate, 2015-07-07:
I had quick play with Resource Hacker and the arduino executable. I opened the .exe file and replaced the icon files with a different one, replaced the arduino icon in the lib folder, and reset the thumbnail and icon cache on Windows.

Basically the windows explorer arduino.exe icon comes from the icons stored into the assembled .exe file, and the taskbar and window icon (the one displayed on the top left corner when the arduino window is opened) come from somewhere else. 

This [post on stack overflow](http://superuser.com/questions/751361/taskbar-icon-for-unpinned-applications-uses-16x16-instead-of-32x32) suggests that:
> Windows uses the Assembly icon when pinned, and the window icon when unpinned

Which seems to be correct in this case, as the arduino.exe pinned icon (for the original or the modified one) is the assembled full resolution icon.

ffissore, 2015-07-08:
Do you have a fix then?

carlosperate, 2015-07-15:
Sorry guys, completely missed the last reply to this issue.
No, I don't have a fix, sounds like a possible bug for launch4j, but I don't really know.

learn-more, 2015-07-17:
In windows you can update both the 'small' and the 'big' icon.
Chances are that (in this case launch4j) does only update one of the 2.
If a window is created, the big icon needs to be set explicitly, if it is not set, it will inherit the small icon.

`
SendMessage(hwnd, WM_SETICON, ICON_SMALL, hIcon);
SendMessage(hwnd, WM_SETICON, ICON_BIG, hIcon);
`
where hwnd is the window, and hIcon the loaded icon.

moritzeto, 2015-07-20:
Before all, thanks to all of you for trying to answer my question.
Hi learn-more, where do you go for changing what you are saying? ( SendMessage(hwnd, WM_SETICON, ICON_SMALL, hIcon); SendMessage(hwnd, WM_SETICON, ICON_BIG, hIcon); )

learn-more, 2015-07-24:
The location where this should be changed is the location that Creates the native windows form, and glues it to java.
I am assuming this is in launch4j?

learn-more, 2015-07-24:
Some further research shows that the frame's icon is manipulated from Arduino, so it should be possible to do this without altering launch4j.

In Base.java, static public setIcon calls frame.setIconImage.
there is another function in a frame, setIconImages (which takes a list of icons instead of one icon).
By passing this a small and a big icon, the correct icon _should_ be chosen for the frame.

pfeerick, 2015-07-26:
After little digging, this is a old issue opened in 2011 in the old google code issue  tracker ([#469](https://code.google.com/p/arduino/issues/detail?id=469)), and there is an apparent fix mentioned [here](https://github.com/eried/Arduino/commit/91c7a2d41d1d02ebea8aea7afdc99cbef9c51d6e). Basically, it is what @learn-more mentioned above, but apparently in a ready to test form.

This [stackoverflow question](http://stackoverflow.com/questions/15004874/use-different-icons-in-jframe-and-windows-taskbar) might give some more insight.

ffissore, 2015-11-24:
Thank you @pfeerick, fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-24 by ffissore

carlosperate, 2015-11-24:
Fantastic! :+1:

eried, 2015-12-19:
This is great!

commit 7dd6e8f5 (2015-11-24): Windows: proper, non blurry, icon displayed. Fixes #3473