arduino/Arduino #3474: parameter "settings.path" in <<installation folder>>/lib/preferences.txt is ignored at startup

cedricfromenstabretagne, 2015-07-07:
**OS** : Windows 8.1
**Arduino version** : 1.6.5

When Arduino starts, it loads `<<installation folder>>/lib/preferences.txt` file.

About `settings.path` parameter from this file :
```
# if you don't want settings to go into "application data" on windows
# and "library" on macosx, set this to the alternate location.
#settings.path=data
```

If I uncomment this parameter and if I set a specific folder to it, **this value is ignored by Arduino**.
See `//##` comments below in code.

**`processing.app.PreferencesData` class** :
```java
 static public void init(File file) {
    if (file != null) {
      preferencesFile = file;
    } else {
//########################################################
//## read path of default "preferences.txt" file
      preferencesFile = BaseNoGui.getSettingsFile(PREFS_FILE); 
//## see explanations in processing.app.BaseNoGui below
//## preferencesFile is loaded by C:\Users\<<login>>\AppData\Roaming\Arduino15\preferences.txt 
//## and "settings.path" parameter is ignored
    }

    try {
      BaseNoGui.getPlatform().fixPrefsFilePermissions(preferencesFile);
    } catch (Exception e) {
      //ignore
    }

    // start by loading the defaults, in case something
    // important was deleted from the user prefs
    try {
      prefs.load(new File(BaseNoGui.getContentFile("lib"), PREFS_FILE));
    } catch (IOException e) {
      BaseNoGui.showError(null, _("Could not read default settings.\n" +
        "You'll need to reinstall Arduino."), e);
    }

    // set some runtime constants (not saved on preferences file)
    File hardwareFolder = BaseNoGui.getHardwareFolder();
    prefs.put("runtime.ide.path", hardwareFolder.getParentFile().getAbsolutePath());
    prefs.put("runtime.ide.version", "" + BaseNoGui.REVISION);

    // clone the hash table
    defaults = new PreferencesMap(prefs);

    if (preferencesFile.exists()) {
      // load the previous preferences file
      try {
        prefs.load(preferencesFile);
      } catch (IOException ex) {
        BaseNoGui.showError(_("Error reading preferences"),
          I18n.format(_("Error reading the preferences file. "
              + "Please delete (or move)\n"
              + "{0} and restart Arduino."),
            preferencesFile.getAbsolutePath()), ex);
      }
    }

    // load the I18n module for internationalization
    try {
      I18n.init(get("editor.languages.current"));
    } catch (MissingResourceException e) {
      I18n.init("en");
      set("editor.languages.current", "en");
    }

    // set some other runtime constants (not saved on preferences file)
    set("runtime.os", PConstants.platformNames[PApplet.platform]);

    fixPreferences();
  }
```

**`processing.app.BaseNoGui` class** :
```java
 static public File getSettingsFile(String filename) {
    return new File(getSettingsFolder(), filename);
  }

  static public File getSettingsFolder() {
    if (getPortableFolder() != null)
      return getPortableFolder();

    File settingsFolder = null;

//########################################################
//## here PreferencesData class is not yet loaded : 
//## so PreferencesData.get("settings.path") returns null value
    String preferencesPath = PreferencesData.get("settings.path");
    if (preferencesPath != null) {
      settingsFolder = absoluteFile(preferencesPath);

    } else {
      try {
        settingsFolder = getPlatform().getSettingsFolder();
//########################################################
//## so settingsFolder is initialized by getPlatform().getSettingsFolder()
//## the value is read in Windows registry :
//## HKCU\\Software\\Microsoft\\Windows\\CurrentVersion\\Explorer\\Shell Folders\AppData
//## there is after a concatenation with Arduino15
//## e.g. : C:\Users\<<login>>\AppData\Roaming\Arduino15
      } catch (Exception e) {
        showError(_("Problem getting data folder"),
                  _("Error getting the Arduino data folder."), e);
      }
    }

    // create the folder if it doesn't exist already
    if (!settingsFolder.exists()) {
      if (!settingsFolder.mkdirs()) {
        showError(_("Settings issues"),
                _("Arduino cannot run because it could not\n" +
                        "create a folder to store your settings."), null);
      }
    }
    return settingsFolder;
  }
```

ffissore, 2015-07-07:
Are you using the IDE from CLI, or just want it to store its stuff to another location?

cedricfromenstabretagne, 2015-07-07:
I just want to store its stuff to another location.

ffissore, 2015-07-07:
I'll work on it. Thanks for the heads up

sinilill, 2015-12-18:
I'm using 1.6.7 and uncommeted the #settings.path=data. Now the library_index.json, package_index.json, package_index.json.sig files are stored in the data folder, but preferences.txt still get's written to the AppData\Local\Arduino15 folder. I think preferences.txt should also be stored in the data folder.

cmaglie, 2015-12-21:
@sinilill 
How can the IDE knows where is the data folder (where to find preferences.txt) without reading the preferences.txt in a know place first?

sinilill, 2015-12-21:
The IDE actually has 2 preferences files, a default preferences file which gets loaded first and a user preferences file. The default preferences file at \lib\preferences.txt contains # DEFAULT PATHS FOR SKETCHBOOK AND SETTINGS. In the default preferences file i have set: settings.path=data.
So settings should be written in the data folder, but it doesn't happen for the user preferences.txt file.
The *.json files are stored in the data folder, but preferences.txt is ignored.

jjdnd, 2018-03-15:
I am using arduino-1.8.5
I am having the same problems as in this thread. I have spent hours trying to get the preferences.txt file to load from my memory stick not the local AppData folder on Windows as I regularly jump between a number of computers, I notice that the topic is 2 years old but I do not see a fix?

matthijskooijman, 2018-09-02:
@jjdnd, your usecase would be supported by the Portable IDE setup, see https://www.arduino.cc/en/Guide/PortableIDE @sobreira, this might be a solution for you as well, depending on what you need exactly.

sobreira, 2018-09-02:
Really appreciate your help.
Worked very fine