arduino/Arduino #3475: IndexOutOfBoundsException when compilation/upload output window updated

Yoda12999, 2015-07-07:
Every once and a while (maybe once every three times) when the compile or upload button is hit, an IndexOutOfBoundsException is created and the requested action does not happen. Generally hitting the button again clears the error and everything works fine.

Error:
```
java.lang.IndexOutOfBoundsException: len=-1
	at sun.font.FontDesignMetrics.charsWidth(FontDesignMetrics.java:501)
	at org.fife.ui.rsyntaxtextarea.TokenImpl.getWidthUpTo(TokenImpl.java:528)
	at org.fife.ui.rsyntaxtextarea.TokenImpl.getWidth(TokenImpl.java:499)
	at org.fife.ui.rsyntaxtextarea.RSyntaxUtilities.getTokenListWidth(RSyntaxUtilities.java:1142)
	at org.fife.ui.rsyntaxtextarea.RSyntaxUtilities.getTokenListWidth(RSyntaxUtilities.java:1121)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.getLineWidth(SyntaxView.java:304)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.calculateLongestLine(SyntaxView.java:110)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.updateMetrics(SyntaxView.java:892)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.setSize(SyntaxView.java:802)
	at javax.swing.plaf.basic.BasicTextUI$RootView.setSize(BasicTextUI.java:1722)
	at javax.swing.plaf.basic.BasicTextUI.getPreferredSize(BasicTextUI.java:912)
	at javax.swing.plaf.basic.BasicTextAreaUI.getPreferredSize(BasicTextAreaUI.java:120)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1662)
	at javax.swing.JTextArea.getPreferredSize(JTextArea.java:619)
	at javax.swing.JViewport.getViewSize(JViewport.java:999)
	at javax.swing.plaf.basic.BasicScrollPaneUI.syncScrollPaneWithViewport(BasicScrollPaneUI.java:278)
	at javax.swing.plaf.basic.BasicScrollPaneUI$Handler.stateChanged(BasicScrollPaneUI.java:1033)
	at javax.swing.JViewport.fireStateChanged(JViewport.java:1369)
	at javax.swing.JViewport.setView(JViewport.java:969)
	at javax.swing.JScrollPane.setViewportView(JScrollPane.java:1007)
	at org.fife.ui.rtextarea.RTextScrollPane.setViewportView(RTextScrollPane.java:253)
	at processing.app.Editor.setCode(Editor.java:1766)
	at processing.app.Sketch.setCurrentCode(Sketch.java:1000)
	at processing.app.Sketch.load(Sketch.java:121)
	at processing.app.Sketch.prepare(Sketch.java:1071)
	at processing.app.Sketch.exportApplet(Sketch.java:1123)
	at processing.app.Sketch.exportApplet(Sketch.java:1113)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2380)
	at java.lang.Thread.run(Thread.java:745)
```

ffissore, 2015-07-07:
I've noticed that too. It's a timing problem. GUI may not be ready yet when some code requests its state. I'm gonna solve this in the coming months. It's not easy. Blindly rewriting some code may automagically solve it

PaulStoffregen, 2015-08-08:
I'm also seeing this bug.  :(

A1RO, 2015-08-27:
I get a similar error on my side with the same behaviour. For me, it only happens when I check "use external editor" in arduino's preferences. When I uncheck this option, it compiles without a single red line.

I hope this helps.

PaulStoffregen, 2015-08-27:
Same here, lots of Java exceptions when Use External Editor is selected.

FrankBoesing, 2015-09-09:
Same error here. But every time a different stacktrace - sometimes the mentioned IndexOutOfBoundsException. Strange. No fun.

CodeFella, 2015-09-14:
Same Problem here when "use external editor" is checked...

But despite of getting a java.lang.IndexOutOfBoundsException i get an java.lang.NullPointerException

<code>
java.lang.NullPointerException
	at org.fife.ui.rsyntaxtextarea.TokenMakerBase.addToken(TokenMakerBase.java:134)
	at processing.app.syntax.SketchTokenMaker.addToken(SketchTokenMaker.java:62)
	at org.fife.ui.rsyntaxtextarea.TokenMakerBase.addToken(TokenMakerBase.java:108)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.addToken(CPlusPlusTokenMaker.java:1855)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.addToken(CPlusPlusTokenMaker.java:1839)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.addToken(CPlusPlusTokenMaker.java:1828)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.yylex(CPlusPlusTokenMaker.java:2195)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.getTokenList(CPlusPlusTokenMaker.java:1901)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.getTokenListForLine(RSyntaxTextArea.java:1699)
	at org.fife.ui.rsyntaxtextarea.folding.CurlyFoldParser.getFolds(CurlyFoldParser.java:123)
	at org.fife.ui.rsyntaxtextarea.folding.FoldManager.reparse(FoldManager.java:583)
	at org.fife.ui.rsyntaxtextarea.folding.FoldManager$Listener.propertyChange(FoldManager.java:705)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:328)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at java.awt.Component.firePropertyChange(Component.java:8422)
	at javax.swing.text.JTextComponent.setDocument(JTextComponent.java:443)
	at org.fife.ui.rtextarea.RTextArea.setDocument(RTextArea.java:1421)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setDocument(RSyntaxTextArea.java:2373)
	at processing.app.syntax.SketchTextArea.switchDocument(SketchTextArea.java:210)
	at processing.app.Editor.setCode(Editor.java:1771)
	at processing.app.Sketch.setCurrentCode(Sketch.java:1000)
	at processing.app.Sketch.load(Sketch.java:121)
	at processing.app.Sketch.prepare(Sketch.java:1071)
	at processing.app.Sketch.exportApplet(Sketch.java:1123)
	at processing.app.Sketch.exportApplet(Sketch.java:1113)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2389)
	at java.lang.Thread.run(Thread.java:745)
</code>

NicoHood, 2015-10-16:
I dont remember that I got this error even once. Maybe I just ignored it. but not so often. However the other issue (you said a side effect of this) https://github.com/arduino/Arduino/issues/3501 drives me crazy. had this 3 times in a row sometimes.

PaulStoffregen, 2016-02-29:
I believe Arduino 1.6.6 & 1.6.7 may have fixed this problem.  I haven't seen it since 1.6.5.

NicoHood, 2016-02-29:
I think so too.

matthijskooijman, 2016-03-07:
I'll go ahead and close this issue, then. If the problem still occurs with 1.6.7, please leave a message so we can reopen.

closed 2016-03-07 by matthijskooijman

MicahStevens, 2016-04-27:
Arduino 1.6.8 on Arch Linux exhibits this issue pretty consistently when trying to compile. 

```
java.lang.IndexOutOfBoundsException: Index: 53, Size: 28
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at processing.app.SketchData.getCode(SketchData.java:226)
	at processing.app.Sketch.setCurrentCode(Sketch.java:986)
	at processing.app.Sketch.load(Sketch.java:107)
	at processing.app.Sketch.prepare(Sketch.java:1068)
	at processing.app.Editor$BuildHandler.run(Editor.java:1961)
	at java.lang.Thread.run(Thread.java:745)
```

matthijskooijman, 2016-04-27:
@MicahStevens it seems this is a completely different problem, that just also happens to raise the same exception. Could you open up a new issue for it? If possible, include the sketch that triggers the problem, since I suspect it doesn't happen on all sketches?