arduino/Arduino #3476: Webpage information not new/updated

WuerfelDev, 2015-07-07:
Here (https://www.arduino.cc/en/Hacking/DFUProgramming8U2) is not mentioned how to handle the Arduino UNO R3.

Because of that I asked in the irc where a very friendly person checked this for me, and he told me that **no resistor** is needed then. Would be nice if someone could update the page for other users.

matthijskooijman, 2015-07-08:
More specifically, the page mentions to add a resistor when not using the Uno SMD, but it turns out the resistor is _only_ needed for the original Uno (R1), and the R2, R3 and SMD all have the resistor already.

agdl, 2015-07-08:
I corrected the statement in "If your board is Arduino UNO R1" if it is ok for you please close the issue

q2dg, 2015-07-08:
I wouldn't close the issue because there's no assembled ICSP pins to reprogram Atmega16u2 on new boards manufactured by Adafruit! 

I've mentioned this fact here https://blog.arduino.cc/2015/07/04/arduino-uno-assembled-in-us-now-available-in-the-store/#comment-277345 (but my last response doesn't appear) and here https://blog.adafruit.com/2015/07/04/arduino-uno-assembled-in-us-now-available-in-the-store-arduino-teamarduinocc/ (but my comment isn't shown).

matthijskooijman, 2015-07-08:
@q2dg That is completely unrelated to this issue, so there is no reason not to close this about DFU programming documentation because you don't like that the ICSP pins aren't assembled by default. If you 
want to discuss this on this issue tracker, please open a new issue (or perhaps motivate  your concerns on the developer list).

@agdl, looks good, thanks. I'm closing this issue.

closed 2015-07-08 by matthijskooijman

q2dg, 2015-07-09:
@matthijskooijman Ok, you're right, sorry. I'll do that, thanks