arduino/Arduino #3478: Under unknown conditions, multi-GB stderr files generated in temp and never deleted

SpenceKonde, 2015-07-08:
I just cleared out ~75GB of data from my temp folder, in the form of files named "stderr.txt" inside a large number of folders with names like "console27501978494037653.tmp"

In one case, a single stderr.txt file was over 18GB in size!

These files, presumably meant to be temporary, are never deleted, and can apparently grow to obscene size. Unfortunately, I did not retain any of the pathological files, nor did I attempt to view them, so I can't provide any information on what was in them (I was focused on getting my computer working again after running out of disk space on my system drive due to these files, and only later figured out that they were related to Arduino). 

The large files appeared to be generated at random over the course of the past year, including both before and after I updated to Arduino 1.6.x. There were dozens of folders, but in most of the folders, the files were small; it was a few seemingly randomly distributed ones that were absolutely massive. Presumably these are created more frequently than I see them, and most of them are removed.

Chris--A, 2015-07-08:
Just had a look at mine and the biggest was 7 MB. I have deleted all mine and I'll keep an eye on it.

If you had the same IDE window open for a very long time, and compile lots of code, there is a possibility a huge file can be generated. Not removing the files when done can be a problem, however all cleanup tools available delete the temp contents.

SpenceKonde, 2015-07-08:
Yeah - I figure it must have gotten into some state where it continually spewed an error message to stderr (and also that it failed to delete them). It's a real shame I didn't keep any of the files so we could see what error...

ffissore, 2015-07-08:
`console*` folders have been removed in a1e5271015b67f2825fe20012a27960f4c2ffaf6 (available in the hourly builds) and temp `build*` and `untitled*` folders get removed on IDE exit (fixed in #2980 available since IDE 1.6.4)

closed 2015-07-08 by ffissore