arduino/Arduino #3479: Poor error description!

bgma, 2015-07-08:
Hey,

I happened to miss one " 
[image]
which got the IDE to stop / loop, it's been about 10 minutes now.
I run on W7 pro 64-bit. sp1, Arduino-1.6.6-nightly-windows_2015-07-03 

/ Sincerely BG

ffissore, 2015-07-08:
Preprocessor fail again

ffissore, 2015-07-08:
Can you share the sketch?

bgma, 2015-07-08:
Hello,
 
I have taken away a lot of code with the same result.
line that is incorrect is annotated with // <This is the line>
I discovered that if you take away the commented lines in the end, you get a completely different message.
 
Date: Wed, 8 Jul 2015 05:12:43 -0700

ffissore, 2015-07-08:
If you attached the sketch to your mail, that won't work. Please paste it into a
```
code block
```
Thanks in advance

bgma, 2015-07-08:
// SerialSpeed parameters
#define SerialSpeed 57600

//RN-XV parameters
#define RNXVSpeed 9600
#define RNXVRxPin 8
#define RNXVTxPin 9

// 328p parameters
#define A328pSpeed 57600
#define A328pRxPin 14
#define A328pTxPin 15

// ---------------------------------------------------------------------//----------------------------------------
#include <SoftwareSerial.h>                                             //
//#include <RCSwitch.h>                                                 //
                                                                        //
#define bgDEBUG                                                         // #ifdef bgDEBUG = On
#define wifiResetPin          A0                                        // RESET of RN-171
#define WIFI_DEFAULT_TIMEOUT 250                                        // Milliseconds 
                                                                        //
SoftwareSerial A328pSerial(A328pRxPin, A328pTxPin);                     // 14,15  
SoftwareSerial wifiSerial( RNXVRxPin , RNXVTxPin );                     // 8,9 $$$  RNXVRxPin connects to RN-171 Pin 2 and RNXVTxPin to RN-171 Pin 3
                                                                        //
char buf[80];                                                           // Data Buffer when reading & writing data
int  intPos = 0;                                                        // Used to point in buf[80]
                                                                        //
// ---------------------------------------------------------------------//----------------------------------------
void setup( ) {                                                         // setup()
    wifiSerial.begin( RNXVSpeed );                                              // 
    if ( wifiBegin( ) != 0) {                                                   // Start wifl
        Serial.println(F("Failed to start wifl"));                              // Error Message
        // WifiTerminal( );                                                        // Start Terminal
    }                                                                           //
}                                                                       // END setup()
// ---------------------------------------------------------------------//----------------------------------------
void loop( ) {                                                          // loop()
                                                                        //
}                                                                       // END loop()
// ---------------------------------------------------------------------//----------------------------------------

// -----------------------------------------------------------------------------//--------------------------------
void bgSetup(void) {                                                            //
    Serial.begin( SerialSpeed );                                                //
                                                                       //
// -----------------------------------------------------------------------------//--------------------------------
    wifiSerial.begin( RNXVSpeed );                                              // 
//    if ( wifiBegin( ) != 0) {                                                   // Start wifl
        Serial.println(F("Failed to start wifl"));                              // Error Message
//    }                                                                           //
// -----------------------------------------------------------------------------//--------------------------------

// -----------------------------------------------------------------------------//--------------------------------
}                                                                               // END setup( )
// -----------------------------------------------------------------------------//--------------------------------

//// -----------------------------------------------------------------------------//--------------------------------
//                                                                                //  
int wifiBegin( ){                                                               // wiflbegin( )

wifiSerial.println(F("set uart mode 0));                                      // <This is the line>
//                                                                                //
//    wifiFlushPort( );
//    return strncmp( buf, "CMD", 3);                                             //
}                                                                               // END wiflbegin( )
//// -----------------------------------------------------------------------------//--------------------------------
//                                                                                //  
//void wifiReadPort( ){                                                           // 
//    intPos = 0;                                                                 //
//    while ( wifiSerial.available( ) ){                                          // 
//        buf[intPos++] = wifiSerial.read();                                      //
//    }                                                                           // 
//    buf[intPos++] = '\0';                                                       //
//}

bgma, 2015-07-08:
it looks like shit... all spaces are gone can I mail it to you?

ffissore, 2015-07-08:
prefix and suffix the code with 3 backticks -> `

ffissore, 2015-07-08:
You can find more info clicking "Markdown supported" link on top of the textarea

bgma, 2015-07-08:
It looks in all cases the same.
in this commentary, I have replaced all the spaces with the underscore 1 to 1
and when you copy back in the IDE and will replace the core with spaces so it is not quite the same

>>>

//_SerialSpeed_parameters
#define_SerialSpeed_57600

//RN-XV_parameters
#define_RNXVSpeed_9600
#define_RNXVRxPin_8
#define_RNXVTxPin_9

//_328p_parameters
#define_A328pSpeed_57600
#define_A328pRxPin_14
#define_A328pTxPin_15

//_---------------------------------------------------------------------//----------------------------------------
#include_<SoftwareSerial.h>_____________________________________________//
//#include_<RCSwitch.h>_________________________________________________//
________________________________________________________________________//
#define_bgDEBUG_________________________________________________________//_#ifdef_bgDEBUG_=_On
#define_wifiResetPin__________A0________________________________________//_RESET_of_RN-171
#define_WIFIxDEFAULTxTIMEOUT_250________________________________________//_Milliseconds_
________________________________________________________________________//
SoftwareSerial_A328pSerial(A328pRxPin,_A328pTxPin);_____________________//_14,15__
SoftwareSerial_wifiSerial(_RNXVRxPin_,_RNXVTxPin_);_____________________//_8,9_$$$__RNXVRxPin_connects_to_RN-171_Pin_2_and_RNXVTxPin_to_RN-171_Pin_3
________________________________________________________________________//
char_buf[80];___________________________________________________________//_Data_Buffer_when_reading_&_writing_data
int__intPos_=_0;________________________________________________________//_Used_to_point_in_buf[80]
________________________________________________________________________//
//_---------------------------------------------------------------------//----------------------------------------
void_setup(_)_{_________________________________________________________//_setup()
____wifiSerial.begin(_RNXVSpeed_);______________________________________________//_
____if_(_wifiBegin(_)_!=_0)_{___________________________________________________//_Start_wifl
________Serial.println(F("Failed_to_start_wifl"));______________________________//_Error_Message
________//_WifiTerminal(_);________________________________________________________//_Start_Terminal
____}___________________________________________________________________________//
}_______________________________________________________________________//_END_setup()
//_---------------------------------------------------------------------//----------------------------------------
void_loop(_)_{__________________________________________________________//_loop()
________________________________________________________________________//
}_______________________________________________________________________//_END_loop()
//_---------------------------------------------------------------------//----------------------------------------

//_-----------------------------------------------------------------------------//--------------------------------
void_bgSetup(void)_{____________________________________________________________//
____Serial.begin(_SerialSpeed_);________________________________________________//
_______________________________________________________________________//
//_-----------------------------------------------------------------------------//--------------------------------
____wifiSerial.begin(_RNXVSpeed_);______________________________________________//_
//____if_(_wifiBegin(_)_!=_0)_{___________________________________________________//_Start_wifl
________Serial.println(F("Failed_to_start_wifl"));______________________________//_Error_Message
//____}___________________________________________________________________________//
//_-----------------------------------------------------------------------------//--------------------------------

//_-----------------------------------------------------------------------------//--------------------------------
}_______________________________________________________________________________//_END_setup(_)
//_-----------------------------------------------------------------------------//--------------------------------

////_-----------------------------------------------------------------------------//--------------------------------
//________________________________________________________________________________//__
int_wifiBegin(_){_______________________________________________________________//_wiflbegin(_)

wifiSerial.println(F("set_uart_mode_0));______________________________________//_<This_is_the_line>
//________________________________________________________________________________//
//____wifiFlushPort(_);
//____return_strncmp(_buf,_"CMD",_3);_____________________________________________//
}_______________________________________________________________________________//_END_wiflbegin(_)
////_-----------------------------------------------------------------------------//--------------------------------
//________________________________________________________________________________//__
//void_wifiReadPort(_){___________________________________________________________//_
//____intPos_=_0;_________________________________________________________________//
//____while_(_wifiSerial.available(_)_){__________________________________________//_
//________buf[intPos++]_=_wifiSerial.read();______________________________________//
//____}___________________________________________________________________________//_
//____buf[intPos++]_=_'\0';_______________________________________________________//
//}_______________________________________________________________________________

>>>

ffissore, 2015-07-08:
I've edited you previous comment

VaclavSal, 2015-07-10:
This is not Arduino.cc forum! 

Please no "publish your full code " , "why you doing that" etc  STUPID time wasting remarks!
Either you are qualified to fix this or not.

The OP went  out of his way to SHOW what fails.  
it DOES NOT MATTER where , why , how etc etc .

It FAILS at F macro - all you have to do just DUPLICATE the line of code ! Duh !

I second his findings, seen same few times - the error output does go crazy.

ffissore, 2015-07-10:
@bgma I've tried compiling your sketch and IDE reports `sketch_jul10a.ino:77:0: error: unterminated argument list invoking macro "F"` which is expected. I guess I pasted a wrong sketch. Please mail me your sketch at f.fissore@arduino.cc

ffissore, 2015-09-30:
Fixed by https://github.com/arduino/arduino-builder available with latest hourly http://www.arduino.cc/en/Main/Software#hourly

closed 2015-09-30 by ffissore

lmihalkovic, 2015-12-06:
[LMCHK]