arduino/Arduino #3480: Change the colour of PREPROCESSOR words (Suggestion)

ReeceM, 2015-07-08:
I was wondering if it would be okay to change the colour of preprocessors to the way C++ or othe IDEs colour them eg.
```C++
#define xyz
#include <xyz.h>
```

I thought this would be better as it makes them more distinguishable from other things as the straight black colour to them just seems ugly and it looks like other code when it is mixed up around other variables(although you wouldn't have stuff clumped together):

[image]

Compared to this:
[image]

I edited the `theme.xml` `PREPROCESSOR` style token and put a new value in for fg, very simple though.

```xml
<style token="PREPROCESSOR" fg="9a4e00" bold="false"/>
```

A link to the [color charts.](http://www.color-hex.com/color/9a4e00)

ffissore, 2015-07-08:
/cc @00alis

00alis, 2015-07-08:
Good catch @ReeceM!

#define and #include should be green because they are part of the Structure category, as you can see in the Language Reference page https://www.arduino.cc/en/Reference/HomePage

We can make this change for the next release! Thanks for your feedback!

ffissore, 2015-07-08:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-07-08 by ffissore

ReeceM, 2015-07-08:
Cool, I'm glad it was of use to point it out and help, its bothered me since I noticed it. 

I did it brown as I liked the look N++ gives for C++ code style.

00alis, 2015-07-08:
Thank you both!! :)

commit 3cf12048 (2015-07-08): Keywords: added preprocessor directives. Fixes #3480