arduino/Arduino #3488: git intergaration

lloyd-g, 2015-07-09:
this is more a question. is there a github integration tool for sketches you to share via github.
started project and it is getting messy.
I'm using Ubuntu.

Chris--A, 2015-07-09:
I just run git alongside my IDE.

If you change the state of a sketch while the IDE is open, simply use the up arrow (open) to re-open the sketch effectively grabbing the changes into the IDE.

lloyd-g, 2015-07-09:
do I have to put my project in the sketch folder?
I started it with web interface and checked out a copy but put it in my Ubuntu home folder.
project is https://github.com/lloyd-g/ArduinoRainStation.
couldn't find wiki for work flow with Arduino?

ffissore, 2015-07-09:
Please don't use github issues for these kind of support questions. They belong to the Arduino Forum http://forum.arduino.cc/
However, my advice is to control the whole sketchbook folder with one git repo: this way you can control even the libraries you use

closed 2015-07-09 by ffissore