arduino/Arduino #3493: [DOC/WEB] Deprecate libraries related to retired boards

q2dg, 2015-07-09:
Reading the information shown in main product page (https://www.arduino.cc/en/Main/Products), I think main libraries page (https://www.arduino.cc/en/Reference/Libraries) should be update. Specifically:

-Remove TFT library from "Standard libraries" section
-Remove "Esplora only library" section
-Remove "Arduino Robot library" section
-Remove Leonardo and Esplora boards from "USB Libraries" section's title 
-Remove entire "Open Accessory for developing with Android" section

On the other hand, I assume TFT, Esplora and Robot libraries will be removed in default installations of future IDE's versions, won't they be?

ffissore, 2015-07-10:
We don't want to stop people using those boards, even if we don't make them anymore.

q2dg, 2015-07-10:
Ok, but it is really confusing to newcomers. 

What I would do then is to add in https://www.arduino.cc/en/Reference/Libraries a section title like "Libraries for retired products" (or similar) and to place this content below:

-The link to TFT library (removing it from "Standard libraries" section)
-The "Esplora only library" section
-The "Arduino Robot library" section
-The entire "Open Accessory for developing with Android" section

I would apply the same idea to the IDE: to add an special category below "Include library" menu (like existing "Arduino libraries" or "Contributed libraries") called "Retired libraries" (or similar) listing TFT, Esplora and Robot.

facchinm, 2015-07-13:
Discussion has moved to the Developers mailing list so closing this one. Reopen if needed

q2dg, 2015-07-13:
Sorry, but I'm not able to see which topic is related in Developers mailing list to this issue. I only requested a slight makeup of "Libraries" web page, that's all

facchinm, 2015-07-14:
Ssorry @q2dg , I thought that your topic about board decisions was a duplicate of this in a different place, my fault.
Reopening and asking @00alis what she thinks about it

00alis, 2015-07-14:
We are currently redesigning https://www.arduino.cc/en/Reference/Libraries, we will definitely make it clear for newcomers that those libraries apply to retired products, thank you for your feedback @q2dg!

I think we should keep them in the IDE thou, a lot of people bought these boards in the past and these libraries are pretty useful.

q2dg, 2015-07-14:
Ok, thanks a lot.
I close the issue, then.

q2dg, 2015-09-29:
Well, I reopen the issue because https://www.arduino.cc/en/Reference/Libraries doesn't still show, as @00alis said, which libraries are related to retired products. I repeat my suggestion written above: in this page should be placed a section titled "Libraries for retired products" (or similar) with this content:

-The link to TFT library (removing it from "Standard libraries" section) *OR point to some compatible screens, like Adafruit's products nº nº 358, nº 802 and nº 2088 , among others*
-The "Esplora only library" section
-The "Arduino Robot library" section

I would apply the same idea to the IDE: to add an special category below "Include library" menu (like existing "Arduino libraries" or "Contributed libraries") called "Retired products libraries" (or similar) listing TFT (I'm in doubt), Esplora and Robot ones. Another (more radical) approach would be putting them off from standard IDE's installation and move them to Library Manager, but I know you don't like this.

Thanks

00alis, 2015-09-30:
@q2dg we have updated the Examples from Libraries page on the website, https://www.arduino.cc/en/Tutorial/LibraryExamples with the categories we agreed on, in the next release of the IDE you will see a similar grouping in the menu @ffissore 

We haven't reviewed the Library Reference yet, but I add @SimonePDA in the loop so he is up-to-date about this change that needs to be made whenever he will tackle this section of the website.

Thanks for your feedback!

q2dg, 2015-09-30:
Ooh,cool, thanks!!

One more question, though,talking about the library list...Temboo and SpacebrewYun libraries are installed by default with the IDE but aren't even mentioned in  https://www.arduino.cc/en/Reference/Libraries . Do you plan to reference them (as you do with Firmata) ? Or maybe it would be a better idea to move them to Library Manager?

Thanks

ffissore, 2015-11-26:
Fixed in 1.6.6

closed 2015-11-26 by ffissore

q2dg, 2016-08-24:
Well, clearly this issue is still NOT resolved (but I can't reopen it I don't know why)
Come on...it's just a minute.
Nowadays this page is a complete mess