arduino/Arduino #3501: [Linux?] Arduino IDE starts with a white GUI

NicoHood, 2015-07-10:
This now happened a few times under Linux (Ubuntu 64 bit and Elementary 64 bit, also based on Ubunutu). I start the IDE via desktop icon or command line and the GUI just doesnt display anything but white. It also wont quit, I have to kill the task via System Monitor.
Noticed this since 1.6.0 or so. Maybe this bug is in earlier version because I started with Ubuntu when IDE 1.6.0 was up to date.

[image]

Luckly it happend when I started the IDE in command line. Here is the output. (nightly of just 15 min ago). (BTW: Java 8 integrated in the IDE works perfect so far. Very good idea!)
```
user@pc:~/Documents/Arduino/arduino-nightly$ sh ./arduino
./arduino: 20: ./arduino: [[: not found
./arduino: 20: ./arduino: ==: not found
./arduino: 20: ./arduino: ==: not found
./arduino: 20: ./arduino: ==: not found
./arduino: 20: ./arduino: ==: not found
Picked up JAVA_TOOL_OPTIONS: 

(java:2260): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(java:2260): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(java:2260): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(java:2260): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(java:2260): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",
Gtk-Message: Failed to load module "canberra-gtk-module"
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HID/examples. Is it pre-1.5?
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HID/dev. Is it pre-1.5?
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HID.wiki/pictures. Is it pre-1.5?
WARNING: Error loading hardware folder /home/user/Documents/Arduino/sketchbook/hardware/HID.wiki
  No valid hardware definitions found in folder HID.wiki.
Board Hoodloader:avr:mtdbu1 doesn't define a 'build.board' preference. Auto-set to: AVR_MTDBU1
Board Hoodloader:avr:hoodloader doesn't define a 'build.board' preference. Auto-set to: AVR_HOODLOADER
Warning: platform.txt from core 'Arduino Hoodloader' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'Arduino Hoodloader' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'Arduino Hoodloader' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'Arduino Hoodloader' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/Hoodloader/dev. Is it pre-1.5?
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HoodLoader2/tools. Is it pre-1.5?
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HoodLoader2/dev. Is it pre-1.5?
Could not find boards.txt in /home/user/Documents/Arduino/sketchbook/hardware/HoodLoader2.wiki/pictures. Is it pre-1.5?
WARNING: Error loading hardware folder /home/user/Documents/Arduino/sketchbook/hardware/HoodLoader2.wiki
  No valid hardware definitions found in folder HoodLoader2.wiki.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Invalid library found in /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki: /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki
Invalid library found in /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki: /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki
Invalid library found in /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki: /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki
Invalid library found in /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki: /home/user/Documents/Arduino/sketchbook/libraries/Nintendo.wiki
^C^C^C^C^C^C^C^C^Z
```

NicoHood, 2015-07-10:
"Funny" addition:
I left the program open (idle) for a while now and it now looks like this. The bugged window from before is still open, and one reopened instance with two windows that stayed idle:

[image]

NicoHood, 2015-07-13:
I had this issue now two times in a row, appears very often.
I saw this log now several times as last info after the task kill:
```
./arduino: line 33:  1015 Killed                  $JAVA -DAPP_DIR="$APPDIR" -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel $SPLASH processing.app.Base "$@"
```

NicoHood, 2015-07-13:
1.6.5 also has this issue. Dont know in which version java got bundled, but I have not installed java manually.

bphermansson, 2015-09-08:
I get this error in Fedora 22 when the system recovers from sleep. The Arduino works as it should but if I leave the computer and the system goes to sleep it spits "Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException" when the computer is active again.

ffissore, 2015-09-08:
This is a side effect of #3475 (or #3475 is a side effect of this). Anyway, as written in https://github.com/arduino/Arduino/issues/3475#issuecomment-119228111, fix is not easy as the bug starts down in the way GUI components were tied with business logic. It could require a substantial rewrite which is hard to schedule atm

NicoHood, 2015-09-08:
I want to note that i get this around every 10th time I start the IDE. Its really annoying, also you have to KILL the task, ending doesnt work.

lmihalkovic, 2016-01-21:
you may want to revisit the editor init/restore code. 
For eg I ended up taking the loading asynchronously so that the editor window always shows cleanly no matter the errors that happen during load. Then if something really bad happens, it is delt with in the editor where it happened (the others are properly and independently restored). The other side effect is that there is a constant time to editor-visible-and-looking-ok. This is particularly useful for sketches with more source files, or on slower platform (say raspberry-pi) where full open cycle may take longer.

NicoHood, 2016-01-21:
You really should share your work. Just commenting that you did it better than it is right now doesnt help. I've seen a lot of great fixes from you, but never a PR. I've also seen similar discussion about this topic, but it gets more and more annoying. Share your work and dont just brag with it please.

lmihalkovic, 2016-01-21:
@NicoHood  There are opensource projects which are 100% the result of the passion of individuals who do what they do on their free time. Other projects, belong to and are clearly driven by businesses, employing salaried software professionals who tightly control where and how these project evolve, in order to keep them in sync with business priorities as well as, when this does not conflict, with end user requests.

Given how this project has more to do with the latter model than the former, it strikes me as rational to offer suggestions that the owners of the project, professionals of the software industry like myself, can review the merrits of, as opposed to throwing code around because I think that it is pretty or even effective.

In this instance a fellow professional made the assessment that altering certain parts of the codebase would be a *substential rewrite* (without further clarifications/quantification). Having made the choice to alter this code, I was offering 1) a short overview of a limited scope rewrite, 2) additional rational for potentially deciding in favor of altering the code (aside from strictly addressing this issue) 3) and a matching time cost, the idea being that the professionals at Arduino LLC could use this information, among many other things, to decide whether or not this was worth them investing the company's resources.

Now, offer any wording that can be used to convey humbly and without hurting anybody's sensibilities, and I will gladly follow the pattern.

NicoHood, 2016-01-21:
Maybe they should hire you ;)