arduino/Arduino #3502: CMD+[ and CMD+] backwards and incorrect on Mac

Jed-SWSDT, 2015-07-10:
At least with version 1.6.5, the functionality of CMD+[ and CMD+] is still not correct. It used to move the entire line regardless of where the cursor was in it. Now that only works for decreasing. For increasing it adds a tab where the cursor is. It was an easy way to change the line indent without having to move the cursor to the beginning (requiring another key combo or mouse move).

Also, they now move the line in the opposite direction from standard. 

Note, this key-combo is used throughout the Mac environment, including Mac Mail (increase/decrease indent level) and shouldn't be confused with tab tab+shift which usually has a different behavior/use.

facchinm, 2015-07-13:
@damellis

damellis, 2015-07-13:
I see the same (incorrect) behavior here, on OS X 10.10.4 w/ Arduino 1.6.5.

Jed-SWSDT, 2015-11-11:
The issue is still present on 1.6.6

damellis, 2015-11-12:
I confirm the problem on Arduino 1.6.6 on Mac (OS X 10.11): the keys are switched (CMD+[ increases the indent instead of decreasing it and vice-versa; and increasing the indent inserts spaces at the cursor, not the start of the line).

damellis, 2015-11-23:
I looked a bit more into this... The first problem is that the open / close actions are simply swapped here: https://github.com/arduino/Arduino/blob/master/app/src/processing/app/syntax/SketchTextAreaDefaultInputMap.java#L48

The KeyEvent.VK_OPEN_BRACKET is a left bracket and should be RSyntaxTextAreaEditorKit.rstaDecreaseIndentAction. For KeyEvent.VK_CLOSE_BRACKET (right bracket), things are more complicated because it looks like RSyntaxTextArea doesn't include an indent action. As a result Editor.handleIndentOutdent() (https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Editor.java#L1867) has to create a text selection before calling the RSyntaxTextArea insertTabAction. If there's no selection, the insertTabAction will actually insert a tab key instead of increasing the indentation of the current line as we'd like it to. So I guess we either need to create a new RSyntaxTextArea action (RSyntaxTextAreaEditorKit.rstaIncreaseIndentAction) or insert a similar hack to create a selection before calling insertTabAction (and remove the selection afterwards).

closed 2015-12-23 by cmaglie

lmihalkovic, 2015-12-27:
[LMNTS]

peabo7, 2015-12-27:
What do your [LM*] messages mean?

commit 480f2fa0 (2015-12-23): Merge branch 'issue3502-indent' of https://github.com/damellis/Arduino

Fix #3502