arduino/Arduino #3503: Unexpected results with --board & --verify arguments in CLI

toddtreece, 2015-07-10:
I've been experimenting with adding Travis CI to test Adafruit library builds, and I'm having an issue running builds against the Due. SoftwareSerial is used in the examples, and there's an `#ifdef` to use **Serial1** on the Due instead of SoftwareSerial. The build fails when I set the `--board` and `--verify` in one shot because it seems like the CLI only partially accepting the new board setting:

```sh
$ arduino --verify --board arduino:sam:arduino_due_x_dbg $PWD/examples/IncomingCall/IncomingCall.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
WARNING: library SoftwareSerial claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [sam] architecture(s).
/usr/local/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial/SoftwareSerial.cpp:45:30: fatal error: util/delay_basic.h: No such file or directory
 #include <util/delay_basic.h>
                              ^
compilation terminated.
Error compiling.
The command "arduino --verify --board arduino:sam:arduino_due_x_dbg $PWD/examples/IncomingCall/IncomingCall.ino" exited with 1.
```

It does work if I set the `--board` and `--save-prefs` first:
```sh
$ arduino --board arduino:sam:arduino_due_x --save-prefs
Picked up JAVA_TOOL_OPTIONS: 
Xlib:  extension "RANDR" missing on display ":1.0".
The command "arduino --board arduino:sam:arduino_due_x --save-prefs" exited with 0.
8.30s$ arduino --verify $PWD/examples/FONAtest/FONAtest.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 26,488 bytes (5%) of program storage space. Maximum is 524,288 bytes.
The command "arduino --verify $PWD/examples/FONAtest/FONAtest.ino" exited with 0.
```

This happens in v1.6.5 on both Linux & OS X.

matthijskooijman, 2015-07-11:
Weird, I'm using this kind of thing all the time for my sketches. Though thinking of it, I only switch between avr-based boards. Looking at your output, it seems --board updates the actual board to use, but not the architecture-specific libraries directory.

What IDE version are you using? Could you try the latest hourly?

Could you add --verbose to see what compiler commands are being used?

toddtreece, 2015-07-13:
@matthijskooijman i'm using 1.6.5 on both Ubuntu (on Travis CI) and OS X. I'll give the latest hourly a try on OS X with `--verbose` and report back.

Semi-related question: Is there a URL for the latest stable version of the IDE? It looks like the hourly build is always `https://www.arduino.cc/download.php?f=/arduino-nightly-macosx.zip`. It would be helpful for testing to have a similar link for the stable version of the IDE so Travis CI configs don't have to be updated after every IDE release.

sandeepmistry, 2015-10-08:
@toddtreece can you please try the latest nightly build? https://github.com/arduino/Arduino/pull/3759 has been merged and resolved #3740 (which looks like a duplicate of this issue).

closed 2015-11-26 by ffissore