arduino/Arduino #3504: polluting output?? ...and this is OK?

VaclavSal, 2015-07-10:
I am resoonding to recent comment about "not polluting the output".
Please explain this to me, 
it is a compiler output of empty sketch.

1. the C:\Documents and Settings\Vaclav\Local Settings\App.... line does not word wrap - it used to 
2. the error stop is reported twice ( not a big issue ) 
3. why are UNUSED invalid libraries reported?

As a user I ,perhaps foolishly,  thought that this forum would appreciate feedback.
I guess I was wrong.  

I am not a developer, only user, and not qualified to provide coding help.
And I have no desire to know what "pull request " is.

#pragma message("Including application CProces class")
#pragma message( "Compiling " __FILE__ )
#pragma message( "Last modified on " __TIMESTAMP__ )
#error STOP
//#include <UTFT.h>
//#include <UTouch.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:

}

Arduino: 1.6.6 Hourly Build 2015/06/30 04:02 (Windows XP), Board: "Arduino Due (Programming Port)"

C:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -IC:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.4\system/libsam -IC:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/CMSIS/Include/ -IC:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/Device/ATMEL/ -IC:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.4\cores\arduino -IC:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.4\variants\arduino_due_x C:\DOCUME~1\Vaclav\LOCALS~1\Temp\build6695816142684938477.tmp\sketch\sketch_jul10a.cpp -o C:\DOCUME~1\Vaclav\LOCALS~1\Temp\build6695816142684938477.tmp\sketch\sketch_jul10a.cpp.o 
sketch_jul10a.ino:6:2: error: #error STOP
sketch_jul10a.ino:2:54: note: #pragma message: Including application CProces class
sketch_jul10a.ino:3:40: note: #pragma message: Compiling sketch_jul10a.ino
sketch_jul10a.ino:4:52: note: #pragma message: Last modified on Fri Jul 10 14:38:29 2015
#error STOP

Invalid library found in C:\Program Files\Arduino1.6.6\arduino-nightly\libraries\UTFT: C:\Program Files\Arduino1.6.6\arduino-nightly\libraries\UTFT
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT
Invalid library found in C:\Program Files\Arduino1.6.6\arduino-nightly\libraries\UTFT: C:\Program Files\Arduino1.6.6\arduino-nightly\libraries\UTFT
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT

matthijskooijman, 2015-07-10:
If you want to respond to #3450, please respond there (you can still comment even though it is closed). Opening new issues just to leave comments clutters github and isn't helpful to follow the discussion. I'm closing this one.

closed 2015-07-10 by matthijskooijman

Avamander, 2015-07-13:
@VaclavSal Could you please get rid of your passive aggressiveness and negative attitude towards people who are making Arduino IDE better. I really do not like reading your comments and issues, you might have had some good ideas but your presentation is and has been awful. (You might also wish to put spaces after commas not before commas)

VaclavSal, 2015-07-13:
Thank you for your comments

PaulStoffregen, 2015-07-13:
I agree.  This negative tone and lack of any meaningful code contribution is a net negative.  Any small value in all this "feedback" is greatly offset by the time wasted of people who actually do contribute code to Arduino.

VaclavSal, 2015-07-13:
I agree too,criticizing my negative tone and correcting my English is a great contribution to resolving the feedback provided. 

As soon as I figure out how to make the compiler output humanly readable again AKA get back to "word wrap" I'll share that.

Avamander, 2015-07-15:
@VaclavSal Could you please stop including the previous comments in your email replies. That is polluting the issue a lot. Invalid libraries are printed because they are invalid and should be fixed, fix em' and they disappear.