arduino/Arduino #3505: Arduino IDE (Wiring / Sketch) for Linux boards or different ARM A5 / A8 boards?

VogonJeltz, 2015-07-11:
hey,
after arduino.cc is supposed to have cancelled it's Arduino Tre project (no longer listed on products page), so regrettably the Due's ARM Cortex M3 is currently the upper limit.

Now, will it be possible to implement the Arduino IDE (Wiring / Sketch) for Linux boards, e.g. on BeagleBone Black or on Raspberry Pi ?
Or maybe even on different, more powerful boards feat. more powerful MCUs than just the Due, e.g. at least an ARM Cortex A5 like on ATSAMA5D3 Xplained evaluation board ?
(see: http://www.cnx-software.com/2014/03/07/79-atmel-atsama5d3-xplained-arduino-compatible-open-source-hardware-board-powered-by-sama5d3-arm-cortex-a5-processor/ )

Chris--A, 2015-07-12:
The ESP8266 is very popular. I use the IDE to program & upload my sketches. It is built around a Xtensa LX106 and can run from 60 to 180 Mhz & has Wifi & 96K RAM.

I got most of mine from $2.50 - $3

VogonJeltz, 2015-07-12:
thank you for your hint!
But the Boards I was referring to, are much more powerful:
the Due has got 92 k RAM - 
but the ATSAMA5D3 Xplained and the BBB have 512MB DDR3 RAM, 
the RasPi 2 even 1GB RAM by 4x 900 MHz cpu clock.
My question was more about boards providing this enhanced level of performance.

VogonJeltz, 2015-08-17:
When can we expect an improvement in the support of bigger and more powerful boards, like e.g.

BeagleBone Black,
Raspberry Pi,
or maybe even for ATSAMA5D3 Xplained evaluation board 

?

per1234, 2017-07-03:
>will it be possible

It already is!
Program Raspberry Pi using the Arduino IDE:
https://github.com/me-no-dev/RasPiArduino

As for the other hardware, of course it's possible. Look around to see if someone's already done it. If not, get working! Of course the Arduino developers aren't going to spend a bunch of time to add support for programming boards Arduino doesn't sell but they have made it extremely easy to easy to create 3rd party hardware packages so we are all free to do what we like with the Arduino IDE.

closed 2017-07-03 by per1234

shiftleftplusone, 2017-07-03:
this is an old topic, it's solved already since quite a long time for the Pi.