arduino/Arduino #3506: [Website] Arduino Playground sidebar edits

per1234, 2015-07-11:
Some suggested edits for the [Playground sidebar](http://playground.arduino.cc/Site/SideBar)(the page has a different password so I am unable to edit it directiy):

- Change **Arduino on other Atmel Chips** to **Arduino on other Chips**
- Change **Suggestions & Bugs** link from the [Google Code issue tracker](http://code.google.com/p/arduino/issues/list) to the [GitHub issue tracker](https://github.com/arduino/Arduino/issues)
- **PARTICIPATE** section:
  - Change **PARTICIPATE** to **Participate**
  - Remove **Suggestions** - this already has a sidebar item(**Suggestions & Bugs**)
  - Remove **Cookbook (addons)**
  - Remove **Drone with Arduino** - I will add this link to the **Exhibition** section once it is removed from the sidebar.
  - Remove **Thermostat with Arduino** - I will add this link to the **Exhibition** section once it is removed from the sidebar.

q2dg, 2015-07-11:
I know it's an utopia, but if someone had the time to erase (very) deprecated and/or incorrect content, it would be perfect.

per1234, 2015-07-11:
@q2dg if you see anything that needs updating or correction feel free to do so. If a page is completely obsolete then you can delete it. The playground is a community project, it's as good as we make it. As of yesterday you don't need to email the webmaster for a password to edit anymore.

q2dg, 2015-07-12:
@perf1234 Ooh, I didn't know now it's possible to edit directly without asking. Thanks. Sorry for not be able to answer your question.

q2dg, 2015-07-13:
Ei!!! Original issue hasn't been resolved! 
Don't close it without any explanation, please!

per1234, 2015-07-15:
@mastrolinux why did you close this issue?

mastrolinux, 2015-07-15:
Because everyone can do that and Playground is not part of the Arduino IDE. It is best to put this info and discussion in the forum in the "website and forum" section. Thanks.

per1234, 2015-07-15:
As I say in the issue, that page has a different password so no everyone can't edit the sidebar. I'm guessing only an admin can do this specific edit. Because of this it seemed more appropriate to put it here in the issue tracker as admins probably don't read every post in the "website and forum" section of the forum.

per1234, 2015-07-30:
It's been 2 weeks since I posted this issue on the forum and no response. @mastrolinux would you please take another look at this issue. I understand that the content of the Playground is the responsibility of the community, but it seems like it's reasonable to expect the Arduino website team to consider making obvious corrections of the Playground infrastructure that can't be edited by the community.

per1234, 2015-09-09:
@ffissore can I ask you a big favor to look at this issue and consider either reopening it, suggesting an appropriate place for me to submit it if this is not the right place for it, or telling me how I can make these changes myself if that's indeed possible. I believe some of these suggestions are obvious and simple improvements on a very important part of the Playground.

mastrolinux, 2015-09-09:
Fixed, closing issue. Thanks a lot.

per1234, 2015-09-09:
@mastrolinux thanks so much for working on this. A couple things though:
>- Change Arduino on other Atmel Chips to Arduino on other Chips

This change wasn't made.
>- Change Suggestions & Bugs link from the Google Code issue tracker to the GitHub issue tracker

This change wasn't made(the link should be changed to https://github.com/arduino/Arduino/issues).
>- Change PARTICIPATE to Participate

There's a typo, you wrote "Partecipate".

per1234, 2015-09-15:
@mastrolinux please read my last comment.

per1234, 2015-09-21:
@mastrolinux please... This is a 2 minute job that has been dragged out over 2.5 months.

per1234, 2015-10-15:
@agdl would you mind taking a look at this issue and consider reopening it? Thanks

matthijskooijman, 2015-11-16:
@mastrolinux, could fix the last few issues reported by @per1234? Or perhaps @agdl can?

mastrolinux, 2015-11-16:
Ok I think I fixed all of those. Can you please check and reopen if necessary?

Thank you

closed 2015-11-16 by mastrolinux

per1234, 2015-11-16:
Looks good. Thanks!