arduino/Arduino #3507: [Library manager] Add Keypad library, please

q2dg, 2015-07-11:
It's hosted in http://playground.arduino.cc/Code/Keypad , but unfortunately author is missing (and library hasn't been updated to follow manager guidelines).

facchinm, 2015-07-13:
Hi @q2dg , I don't believe we will add this to the libraries maintained by Arduino, however being the code GPL you can always fork it, fix the library manager compatibility issues, submit a request for inclusion by yourself and we'll be glad to include it :wink:

Chris--A, 2015-07-13:
I have one here we can use. https://github.com/Chris--A/Keypad

I'm just going to update it for the new 1.5 specification first. Older IDE's can use the playground link.

Chris--A, 2015-07-13:
Ok, it looks good to go!

q2dg, 2015-07-13:
Yeah, thanks!!

matthijskooijman, 2015-07-13:
@Chris--A, perhaps you should add a notice to the playground page, pointing people to your github and the library manager (probably best after it is added to the library manager, of course)?

Chris--A, 2015-07-14:
Yes, once everything is rolling.

cmaglie, 2015-07-15:
@Chris--A 
the library is still missing tags: https://github.com/Chris--A/Keypad/releases
please add one so I can publish into the lib manager

Chris--A, 2015-07-15:
Its added now, sorry.

cmaglie, 2015-07-15:
No problem, it should be online now!

closed 2015-07-15 by cmaglie