arduino/Arduino #3509: [Manual] Can't find reference/Guide_Windows.html and reference/Guide_MacOSX.html in README.md

HirotakaTakayama, 2015-07-12:
README.md needs to be updated. 
Because I can't find reference/Guide_MacOSX.html file on README.md. 
Is the installation page "https://github.com/arduino/Arduino/wiki/Building-Arduino"?

This text should be changed:

    Detailed instructions are in reference/Guide_Windows.html and reference/Guide_MacOSX.html. 
    For Linux, see the Arduino playground: http://www.arduino.cc/playground/Learning/Linux

I recommend replacing it with this:

    Detailed instructions are in https://github.com/arduino/Arduino/wiki/Building-Arduino

agdl, 2015-09-01:
this is a duplicate of https://github.com/arduino/Arduino/issues/2596

closed 2015-09-01 by agdl

sinisterrook, 2015-11-22:
Seems to be still relevant