arduino/Arduino #3510: 1.6.6 cannot read temporary .hex file when uploading sketch

NicoHood, 2015-07-12:
Linux 64 bit, hourly built 2015/07/10 01:26

I get this error when I try to upload with an external .boards file via hardware folder. For example a copy of the Arduino Uno does not work, while the original Arduino Uno works.

I used a different core, not the Arduino core. Not sure if this affects it. I think this issue is related to this PR where you change the compile folders/sturcture.

I do not remember to have this in 1.6.5. I cannot test again since I have no java8 here and to avoid problems I will use the bundled java8

```
avrdude: Device signature = 0x1e958a
avrdude: reading input file "/tmp/build2823025267484140062.tmp/sketch/HoodLoader2_RunBootloader.cpp.hex"
avrdude: can't open input file /tmp/build2823025267484140062.tmp/sketch/HoodLoader2_RunBootloader.cpp.hex: No such file or directory
avrdude: read from file '/tmp/build2823025267484140062.tmp/sketch/HoodLoader2_RunBootloader.cpp.hex' failed

avrdude done.  Thank you.
```

NicoHood, 2015-07-12:
https://github.com/arduino/Arduino/commit/e097439eab6be8b63f786276cec87568044a3543#diff-0d45b5349ab1c852b7cad3e6b173e880
https://github.com/arduino/Arduino/commit/5f31f4b8ddfdd601e397304f2e4d19c41de693d0#diff-0d45b5349ab1c852b7cad3e6b173e880

Updated my platforms.txt and now works. I think this might be a problem for 3rd party libraries. You should add a note that the maintainer should update those files or you add a link for the old style.

facchinm, 2015-07-13:
@NicoHood thanks for reporting; however the path change only affects custom cores maintainers so if you could post here the your diff we'll add a note (with examples) in the 3rd party cores wiki page.

NicoHood, 2015-07-13:
Well I copied the platforms.txt from the arduino folder to my core.

I still have some probs with my git and the github server is currently heavy loaded, but this must be the fix:
https://github.com/NicoHood/HID/commit/6c8cb0a08bee0d643c0c90cffb0dfafdb47a4446

agdl, 2015-07-16:
@NicoHood I added this statement

Note for core maintainers: In IDE 1.6.6 or greater the folder structure is different so maintainers should update their platform.txt according to this fix made by NicoHood

https://github.com/NicoHood/HID/commit/6c8cb0a08bee0d643c0c90cffb0dfafdb47a4446

In this page
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

Let me know if it is good for you!

NicoHood, 2015-07-16:
Looking good. :)

agdl, 2015-07-16:
@NicoHood can this issue be closed?

closed 2015-07-16 by NicoHood