arduino/Arduino #3511: Fails to start when sketchbook does not exist

matthijskooijman, 2015-07-12:
When a non-existent sketchbook folder is specified, the IDE gives a popup saying "Sketchbook folder disappeared", saying it will revert to the default sketchbook folder. However, instead of doing that, it just quits, without opening up the IDE interface.

To reproduce:

    arduino --pref sketchbook.path=/nonexistent

matthijskooijman, 2015-07-12:
Seems I drew conclusions too soon: the problem is actually that if --pref is given, no interface is shown at all (as if e.g. --verify was given). I'm not entirely sure how this stuff was supposed to work, but I don't think this is really intentional.

zephyrr, 2015-12-01:
Since the man pages says that the IDE will not be opened if there are any command line parameters, I don't know if this is really a bug, or a feature request.  But it would be very handy and perhaps simple to do.  Especially if we accept that a pref set via the command line when opening the GUI IDE would be saved as permanent.

Simplest description would be to bring up the IDE even if there were cmd line params, UNLESS --verify or --upload is given.  Or add a cmd line param to explicitly control whether the IDE is used/suppressed.

My use case is some code whose author requires a few tweaked libraries; the standard versions will not work with it, and the tweaked libraries would break some normal code.  This would be elegantly fixed by having two sketchbooks (and user libraries beneath them) which can be invoked via two Windows shortcuts (each having different cmd line params).  Way more elegant than keeping two full installations of Arduino in different folder trees, just to allow each to have a different sketchbook - the current approach by users of that software.

Actually I'd prefer that there be an option not to save the cmd line pref as permanent, but I can live with it being made permanent if that makes it easier to implement.

per1234, 2017-07-03:
How about this?:
- If only "configuration" options such as `--pref`, `--port`, `--board`, `--verbose`, `--verbose-build`, `--verbose-upload` are passed then start the GUI. Otherwise these options by themselves are pointless. This should have no impact on headless use.
- If any other "action" options are also passed, including `--save-prefs` then don't start the GUI.

pfeerick, 2017-10-21:
Is anything happening with this issue? It's a bit annoying having to have the IDE run twice when using the command line prefs stuff to load different multiple sketchbooks and board from different shortcuts. (I use it to have architecture specific library and sketch combinations - so there are no clashes).