arduino/Arduino #3512: Make it possible to edit libraries in the IDE

probonopd, 2015-07-12:
Books on Arduino give the advice that "Because the Arduino IDE is set up to recognize .ino files, it is often easier to write the header and source code files using a simple text editor like Notepad" (Jack Purdum, Beginning C for Arduino, Second Edition).

This is a shame.
The Arduino IDE should be able to open libraries for editing in a sane way.

per1234, 2015-07-12:
I also think this would be very nice. I really like the IDE and find it easiest just to use one program for editing all my Arduino code. I have hacked together a system that allows me to edit libraries with the IDE. I'm guessing that the IDE doesn't allow saving to the library folders so that people can't accidentally modify the examples but it would be nice if the IDE was changed to only write protect the **examples** folder. Then you would just need to put a blank .ino file in the library folder to be able to edit. I also have my system set up so I can edit the keywords.txt and readme.md files as tabs in the IDE.

Chris--A, 2015-07-13:
@per1234 
As far as the file access goes, I would rather have control of it through the OS. If I select the examples folder and remove the read-only property from it and its contents, the IDE should not prevent me from overwriting the files.

per1234, 2015-07-13:
@Chris--A I agree. I need to be able to edit the examples while I'm working on a library and it would be nice to be able to edit them in the examples subfolder of the library folder. How do you see this being implemented? Would the IDE set the files read-only after the library is installed and then leave the attribute alone after that?

PaulStoffregen, 2015-07-13:
Eventually, in a distant future where the Arduino IDE supports breakpoint debugging, display of the libraries and core library files will be needed.  This might be a good time to think about user interface/experience trade-offs related to displaying so many files, perhaps only when the user needs them.

per1234, 2015-07-13:
@PaulStoffregen I agree that the best case scenario is to be able to open an example or test sketch and then have the **option** to open any of the included libraries files as tabs and be able to edit the library code and compile. That might be pretty distant future stuff, and maybe Arduino would consider it too advanced/confusing for the target user.

A more immediately obtainable, albeit less useful, option is just to allow the IDE to save to the library folders. Of course you can't compile them with the IDE, it's just a text editor. I prefer this to using the Arduino IDE for sketches only and then needing a separate program to edit libraries. This option wouldn't require any changes to the user interface or documentation. It would just provide another usage capability for people who want it.

I think making Arduino library development more accessible would benefit users, by allowing easier code reuse, and the community, by providing more available libraries.

ffissore, 2015-08-05:
With 59dfede5acebc544d6d93c33a31d314119896ead examples of libraries with a `.development` flag file in their root folder will not be flagged as read-only. Such file MUST NOT be pushed to git: the hourly job that fetches contributed libraries releases and publishes them to Library Manager will SKIP libs containing that file

per1234, 2015-08-06:
@ffissore thanks for adding this functionality! I noticed this only works with the 1.5 library folder structure. Would it be possible to make it compatible with the 1.0 libraries folder structure also?

per1234, 2015-08-06:
I was wrong, it just requires a `library.properties` file to be present so that's no problem just to add that to any library that's missing it. Thanks again!

lmihalkovic, 2015-10-31:
By making some changes to a couple of core classes below the editor, It is possible to reuse all the existing code and make it work on libs. For the moment i only did it for libraries added locally into a sketch project (not touching the ones in Arduino/libraries for the moment). But it shows theres space for some improvememnt without very much work.
https://github.com/arduino/Arduino/issues/4050

lmihalkovic, 2015-11-26:
Done

carlosperate, 2015-11-26:
It would be nice to have a menu option to "select/deselect development mode", maybe a "development" entry with a tick that is only selectable with libraries?

lmihalkovic, 2016-01-21:
Read-only for system libraries (need padlock icon for status bar)
[image]
Nice to have: submit lib directly from ide via [Publish To Arduino] button...

sgparry, 2016-03-19:
THIS DOES NOT WORK. I am running 1.6.8 x64 on Ubuntu. I have put .development flags in just about every folder I can think of. It neither lets me edit the library source code nor the example sketches for my user created libraries. The code is clear as mud on this. I cannot see why it is locking me in read-only.

Why have this stupid 'feature'? What is the point? It protects _nobody_ and annoys everyone. Some of us actually want contribute code to the community without having to copy our files backwards and forwards or switch editors every time we want to fix bugs or add enhancements to own library code.

tofrnr, 2017-06-13:
It must be possible to open standard C files from where ever they may come (e.g., mostly from other library code but optionally also from other C, C++, or ino sources ), then to store the files as .c or .h files at a different place and then #include them into proprietary .ino code. 

That would not compellingly even implicate to create entire "libraries", and not even to edit entire existing libraries, neither in the sense of standard C (.so, .a) nor of the way how Arduino constructs them out of other files in proprietary subdirectories plus extra properties and keywords files.

But at least the option to simply  open and save .c and .h files must be provided!

dunk8888, 2018-02-05:
.development  doesnt work for me eather,it says must type a file name!!! does it have to be a text file??

dunk8888, 2018-02-05:
Dont know why they dont add an option to put the examples back to default settings or somthing like that so if people do break some code it can be restored,is plainly an option we all want.

sgparry, 2020-08-01:
Just discovered an annoying new drawback to my workaround for editing libraries - the examples scanning in the IDE includes *any* .ino files - so all my empty src.ino files that allow me edit library code directly in the IDE also generate spurious examples in the menu. Why? why is this so dumb? examples go in the examples folder - that's what the spec says.

per1234, 2020-08-01:
@sgparry there is discussion about enforcing the [Arduino library specification regarding the `examples` folder](https://arduino.github.io/arduino-cli/latest/library-specification/#library-examples) here: https://github.com/arduino/Arduino/issues/10235

sgparry, 2020-08-01:
Thanks @per1234 - just found that myself.