arduino/Arduino #3513: Comment/Uncomment broken

probonopd, 2015-07-12:
Steps to reproduce:
* Open a new sketch,  it has a `// put your setup code here, to run once:`line
* Put the cursor into that line
* Select "Comment/Uncomment" from the menu

Expected result:
`put your setup code here, to run once:`
Actual result:
`//  // put your setup code here, to run once:`

Also it would be nice if Comment/Uncomment and Auto Format would work nicely together. Right now, using Comment/Uncomment, then Auto Format, and then Comment/Uncomment again breaks it.

per1234, 2015-07-12:
It only happens with indented comments.

probonopd, 2015-07-12:
Correct. It also happens that Auto Format indents comments.

ffissore, 2015-09-08:
Fixed. Autoformat keeps on formatting comments. This can be easily disabled editing file formatter.conf and saving a copy into your own Arduino15 folder

commit 96e0ee2a (2015-09-08): Editor is now able to find a commented line even if // was not written at its beginning. Fixes #3513
closed 2015-09-08 by ffissore