arduino/Arduino #3514: Spelling in the menus

probonopd, 2015-07-12:
It would be nice if spelling in the menus could be more consistent. We have "Comment/Uncomment" with no spaces around the "/" but "Verify / Compile". Please remove the extra spaces. This is also the case in other languages, e.g., German. The Apple standard for "/" in the menus is without spaces.

I am using [image] on OS X 10.9

facchinm, 2015-07-13:
@00alis

00alis, 2015-07-13:
let's do it, consistency it is always better :)

ffissore, 2015-08-06:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 4d932327 (2015-08-06): Verify / Compile -> Verify/Compile. Fixes #3514
closed 2015-08-06 by ffissore