arduino/Arduino #3515: Error on Arduino CLI opens an error message window.

carlosperate, 2015-07-12:
Tested on Windows 8.1 x64 bits using the `arduino_debug.exe` hourly build `1.6.6 2015/07/10 12:34`.

This happened on exit code 2, when the sketch cannot be found, and it raises an error message window. Execution stops until the window is dismissed (which could be a problem for headless interfaces). No other message is printed on the command line interface stdout or stderr to indicate this was the reason for the failure, although the exit code itself does indicate this.

I also tried to see if the same would happened on a upload error (disconnecting the board on upload), but in that case no window is opened, and works as expected.

facchinm, 2015-07-13:
Hi @carlosperate , thanks for reporting! The issue will be tackled by @ffissore as soon as he is back!

matthijskooijman, 2015-07-13:
This is related to #1981, but it might be easier to fix just the error message first, without removing the GUI requirement altogether.

carlosperate, 2015-07-13:
Yep, it is indeed related to #1981, but I think this case might require special consideration since people working around the X11 issue by creating a virtual display might still have the problem of this window blocking execution until it is dismissed.

ffissore, 2015-09-25:
As far as compilation is concerned, don't use the IDE any more. Use [`arduino-builder`](https://github.com/arduino/arduino-builder) which does not require X11

carlosperate, 2015-09-25:
I am not too familiar with the new arduino-builder, but wouldn't we need to specify in the command line interface a lot more parameters (location of hw libraries and similar)? Is the a simple method to use it like the Arduino IDE CLI?

ffissore, 2015-09-25:
Yes, it needs more params. More power, more params :)

ffissore, 2015-09-25:
Speaking of power, it is twice as fast as using plain `arduino`. Halved our CI server execution times

carlosperate, 2015-09-26:
I definitely like the idea of faster compilation, but more parameters also means that it needs a lot more effort for other applications to interface with it and maintain compatibility with different versions of the IDE. It looks like the compilation and program uploading components are currently being refactored to be able to integrate them into the upcoming Cloud IDE, which is great, but the current desktop IDE CLI was also great due to its simple interface.

Are there any plans to further refactor the compilation and uploading code outside of the java application? Ideally we could have a "arduino_cli" executable taking the same parameters that the current Arduino desktop IDE CLI takes, and have it interface with arduino-builder and avrdude. This program could then be invoked by the launch4j application or any other third party app.

ffissore, 2015-11-23:
We have no such plans

closed 2015-11-23 by ffissore