arduino/Arduino #3516: Add Library - Quirkbot

paulobarcelos, 2015-07-12:
Please add the Quirkbot library to the public index. It conforms to the 1.5 format.
https://github.com/Quirkbot/QuirkbotArduinoLibrary

facchinm, 2015-07-13:
Hi @paulobarcelos, @ffissore is in charge for library manager so expect some delays in integration. Don't worry though, it will be added as soon as he is back!

paulobarcelos, 2015-07-13:
Great! Looking forward to it.

Are there any plans on making this process automated? I can imagine that manually going through these libraries submissions is very time consuming...

facchinm, 2015-07-13:
Testing the conformity of a library with the specifications can be almost fully automated, however there are still corner cases which would need human intervention, so it's unlikely to see a fully automated process in a short time :wink:

cmaglie, 2015-07-15:
The library has been added! Thanks!

closed 2015-07-15 by cmaglie