arduino/Arduino #3517: Add Library - Dallas Temperature

PaoloP74, 2015-07-13:
Please add the Dallas Temperature library to the public index.
https://github.com/milesburton/Arduino-Temperature-Control-Library

facchinm, 2015-07-13:
Hi @PaoloP74 , @ffissore is in charge for library manager so expect some delays in integration. Don't worry though, it will be added as soon as he is back!

PaoloP74, 2015-07-13:
Ok.

cmaglie, 2015-07-15:
@PaoloP74 you must tag a release on the repository to get it published:

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

also a `library.properties` file is missing.

cmaglie, 2015-07-17:
The library has been added.
Thanks!

closed 2015-07-17 by cmaglie

PaoloP74, 2015-09-22:
Latest release is 3.7.5, but Library Manager install version 3.7.2. Bug?

ffissore, 2015-09-22:
I see 3.7.5 listed in http://downloads.arduino.cc/libraries/library_index.json

ffissore, 2015-09-22:
Please open a new issue if you experience a bug. Comments to closed issues may get lost (too many emails)