arduino/Arduino #3518: Add Library - TVout

Avamander, 2015-07-13:
Please add the TVout library to public index.
https://github.com/Avamander/arduino-tvout/

facchinm, 2015-07-13:
Hi @Avamander, as @ffissore is in charge of library inclusion expect some delays but don't worry, it will be added soon!

cmaglie, 2015-07-15:
@Avamander 

`sentence` and `url` fields must be defined in `library.properties`, please add them a re-tag the new revision.

cmaglie, 2015-07-15:
In particular the `url` field is missing in the revision that you tagged as 1.0.0:

https://github.com/Avamander/arduino-tvout/tree/v1.0.0

Avamander, 2015-07-15:
Fixed.

cmaglie, 2015-07-15:
ok, it should be available now. Thanks!

closed 2015-07-15 by cmaglie

Avamander, 2015-07-15:
Confirmed, library is indeed downloadable directly from IDE.

abcbarryn, 2015-10-12:
I have an updated version that works with the Arduino Leonardo as well as other previous Arduinos, improved wiki pages, and a home web page at http://abcbarryn.github.io/arduino-tvout/
What is the process to make this newer version available for direct downloading?

I have added Avamander and cbmeeks both as collaborators on the new site.

To do: Clean up redundant directory structure.

NicoHood, 2015-10-12:
You should create a Pullup request that the original owner can review (not on the oroject you forked) and then the owner needs to update the library.properties and after a few hours the new version will appear.

Seems there are two versions. The one you forked and this one (from the author who started this issue)
https://github.com/Avamander/arduino-tvout/

You should better open the PR to the project linked above.

abcbarryn, 2015-10-12:
Done, however his web site still lacks the following that my site has:
A Wiki with help pages.
A Web page.

Just FYI

At this point the code should be the same.

I actually had to create another fork to submit the pull request.

NicoHood, 2015-10-12:
Github pages are nice - mostly noone uses them.
Github Wikis are perfect - if you place a big link in the readme.
Otherwise a readme with simple information is mostly enough if the project isnt too complex. Self explaining examples are even more important.

Your PR seems to be at the right place now. If noone answers you should probably contact the author direct. (Because the github repo is 5 years old, but that doesnt have to mean anything).

Avamander, 2015-10-12:
Should I do something? I just rehosted the original library and made it compatible with the IDE's library manager, even the website is from the original repository.

NicoHood, 2015-10-12:
Maybe you should contact with the original author and ask him to integrate the changes in its github repo. If he is not answering then your solution is okay, but I would not like it if someone copies my repository just because its 5 years old. Its open source for sure, but the problem is when people are wondering what version is the most up to date now. One of the repos should be shut down or at least link to ther other in my opinion. What did the author of the original lib say to this?

abcbarryn, 2015-10-12:
FYI: I updated the URL of my page to http://abcbarryn.github.io/TVout/

Avamander, 2015-10-12:
@NicoHood The original repo was on Google Code, if I remember correctly I also emailed the owner and got no response? The repo itself was also really abandoned. The repo's licence also allows this if I understand it correctly.

Avamander, 2015-10-12:
@abcbarryn Shall you delete your repo now that your changes are in my repo? (I also added you as a collaborator)

NicoHood, 2015-10-12:
Oh that could be true. I just saw that @abcbarryn forked it from here:
https://github.com/cbmeeks/arduino-tvout

I dont know if this is the original author. Open an issue and ask him there, thats the best idea.
@abcbarryn does not need to delete its repo, because its no repo. Its a fork which is totally okay.
I also did not want to offend you, just asking if you asked the author (cause normally one would fork the original repo and not just reupload it). Its good that someone takes care of those older "forgotten" libraries.

Avamander, 2015-10-12:
Just saying that the original author is Myles Metzler. The original repo is not on Github, it is on Google Code. My repo is actually a copy hosted on Gihub, not a reupload as you can see commit history goes back to 2010.

abcbarryn, 2015-10-13:
I'm glad to see you are still involved with this. I can delete or repoint my repo to yours, but can I migrate the webpage and wiki I created at:
http://abcbarryn.github.io/TVout/
and
https://github.com/abcbarryn/TVout/wiki

I haven't copied all the pages from the old Google Code pages yet, but I have copied a lot of them, and I redirect to the rest.

abcbarryn, 2015-10-13:
I do think we should agree on on place to host this for the updated version if possible, or at least fewer copies. :) The way git hub forces you to make a copy to propose changes is awkward. I was thinking of either setting up a link on the web page for people to request access, or accepting and merging in diff files or other patches. Another question… What to do about TVoutfonts? Should that be handled as a separate lib or integrated into TVout? Right now, I had to copy it's folder to the library folder manually to get it to work. Also, should some of the other .h files get moved to a sub folder? Because when I add the library it seems to add includes for all the .h files, even the ones TVout.h already includes. It works anyway, but it's messy.

matthijskooijman, 2015-10-13:
As far as Arduino is concerned, https://github.com/Avamander/arduino-tvout/ is now the "official" location for this library, so merging any improvements to that repository would be the obvious way forward. Alternatively, if want to move maintainership to someone else, you can let the Arduino team know and the library list can be updated. This would normally require consent from the current maintainer. Reading the discussion more closely, I think the code changes have already been merged into Avamander's repository, only the wiki and gh-pages stuff is left. AFAIK both can actually be accessed as a git repository as well, which would mean that merging stuff into Avamander's repository should be fairly easy (i.e. it shouldn't require manual copy-pasting). After these docs are merged, it might make sense to delete @abcbarryn's fork (and perhaps re-create a clean fork), to prevent confusion about what is original version and what is the fork, but that's up to you.

Avamander, 2015-10-14:
@abcbarryn 
TVoutfonts should be kept as it is. But examples should be updated so that user knows that that should be done too. 

Wiki and everything can also be migrated to my repo.