arduino/Arduino #3520: IDE removes the TAB!

bgma, 2015-07-13:
Hey,

I try to put in 2 TAB and } one after the other in a line.
2 TAB is going well but when I press } it removes 1 TAB away.

I know it is missing 1 { but I will in any case be able to format my code without the IDE is doing something on it's own.
In the run on W7 Pro 64-bit. sp1, Arduino-1.6.6-nightly-windows_2015-07-10 12:34

/Best regards BG

facchinm, 2015-07-13:
Hi @bgma , the new Editor contains some "automation" which has been considered helpful for a lot of users (like parenthesis matching). However, as soon as @ffissore is back, he'll be able to give you a proper answer :wink:

ffissore, 2015-09-25:
@bgma that happens because the editor is aligning the closing curly brace to the one you opened above. Either open the second curly brace or first type the closing one, then indent it.

closed 2015-09-25 by ffissore

bgma, 2015-09-25:
is it possible to turn off this BUG? or must i use notepad to get rid of it?

ffissore, 2015-09-25:
I'm sorry, it's not possible