arduino/Arduino #3521: ArduinoToBreadboard tutorial hardware configuration archive is not compatible with 1.6.5

per1234, 2015-07-13:
The hardware configuration archive [Breadboard1-5-x.zip](https://www.arduino.cc/en/uploads/Tutorial/Breadboard1-5-x.zip) linked in the **Minimal Circuit** section of www.arduino.cc/en/Tutorial/ArduinoToBreadboard doesn't work with Arduino IDE 1.6.5r2 etc.

The following changes need to be made to the archive:
- Add to **boards.txt**:
  - `atmega328bb.bootloader.tool=arduino:avrdude`
  - `atmega328bb.upload.tool=arduino:avrdude`
- Remove the line `atmega328bb.bootloader.path=arduino:atmega` from **boards.txt**.
- It doesn't include a bootloader file so either:
  - https://github.com/arduino/Arduino/issues/2656 needs to be fixed so it can use the arduino core bootloader file and update the bootloader.file line to: `atmega328bb.bootloader.file=arduino:atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex`
  - **or** add the bootloader file **ATmegaBOOT_168_atmega328_pro_8MHz.hex** to the bootloaders subfolder of the archive.

agdl, 2015-07-16:
@per1234 there is a new archive now that works with 1.6.5. I included the bootloader file in the archive. If it is good for you please close this issue!

per1234, 2015-07-16:
@agdl the url for breadboard-1.6-x.zip (https://www.arduino.cc/en/6-x/Zip?action=edit) on the tutorial page doesn't work

agdl, 2015-07-16:
i just noticed it. Try in a minute. updated it

per1234, 2015-07-16:
Still doesn't work: https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard?action=upload&upname=breadboard-1.6-x.zip

agdl, 2015-07-16:
try to refresh the page cleaning the cache please

agdl, 2015-07-16:
ok it works for sre now. It is online sorry about this

per1234, 2015-07-16:
Works great, thank you!

closed 2015-07-16 by per1234