arduino/Arduino #3526: Hardware-SPI libs  for TFT displays as a substitute for UTFT?

VogonJeltz, 2015-07-14:
would it be possible to add Hardware-SPI libs for TFT displays, working also with the Due ?
I'm using Henning Karlsen UTFT libs so far, but unfortunately they don't work on the SPI-Header simultaneously with Hardware-SPI device libs, e.g. <SD.h>.

E.g., without a SD card on the bus I can even configure and assign  the SPI pins (on Mega: 50-52, on Due: 74-76) but as soon as I'm adding a SD card to the bus the TFT don't work (e.g., having CS_SD=38, CS_TFT=52).

So I always have to use extra different MISO and SCK pins for TFT and SD which finally wastes pins unnecessarily.

VogonJeltz, 2015-07-17:
ps,
this is one of the TFT displays which actually don't work with additional hardware SPI devices on the SPI header:
http://www.mindstormsforum.de/viewtopic.php?f=78&t=8491#p66123

agdl, 2016-07-12:
This issue was moved to arduino-libraries/TFT#4

closed 2016-07-12 by agdl