arduino/Arduino #3527: Uploading (CTRL+U) restarts Serial Monitor on compile error

NicoHood, 2015-07-14:
If you want to upload a program and the compiling fails it will reset the Serial Monitor. I think it should only do it with a proper uploading so you can control the program flow better. Depending on what you are using the Arduino this can cause unexpected problems (restart of the old, buggy program).

facchinm, 2015-07-15:
Hi Nico, I think you are right; the issue will be tackled by @ffissore as soon as he is back

ffissore, 2015-08-03:
What do you mean with "resetting". AFAIK, its contents are not cleared

matthijskooijman, 2015-08-03:
I suspect it means closing/disableing and re-enabling/re-opening, which would cause the board to reset?

NicoHood, 2015-08-03:
restart the Arduino (while Serial monitor is still opened)

ffissore, 2015-09-09:
If I got this right, you're saying that you don't want serial monitor to suspend when you have it open and you try to upload a sketch that doesn't compile. Because of the way this behaviour is coded, it's quite hard to fix at the moment. I'll leave this open as an improvement suggestion. Thanks!

NicoHood, 2015-09-09:
Yes. It should just reset when it really tries to upload. Meaning the reset should occur after a compilation, not before.

per1234, 2021-04-07:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8146. Even though this one came first, the other has a little bit more discussion.

closed 2021-04-07 by per1234