arduino/Arduino #3528: [Library Manager] Please add the following libraries

nullboundary, 2015-07-15:
Everything should checkout as 1.5+ compliant. Let me know if there are any problems. 

https://github.com/nullboundary/AtTouch
Arduino Library for the Atmel AT42QT1070 Touch Capacitance Chip 

https://github.com/nullboundary/SST25VF
Arduino Library for controlling the SST Nor Serial Flash SST25VF family

https://github.com/n0m1/Sleep_n0m1
A library that sets the Arduino into sleep mode for a specified length of time, or until an interrupt

https://github.com/n0m1/Comp6DOF_n0m1
An Arduino Library for compass tilt compensation and hard iron offset

https://github.com/n0m1/MMA8453_n0m1
an Arduino Library for the Freescale MMA8453Q & MMA8452Q Accelerometer

facchinm, 2015-07-15:
Hi @nullboundary , your libraries will be included as soon as @ffissore is back, expect some delay but don't worry!

cmaglie, 2015-07-15:
Hi, @nullboundary 

there are some problems:

AtTouch: you should tag the revision that has a library.properties file, please note that the tag 1.0.3 doesn't have a library.properties at all: https://github.com/nullboundary/AtTouch/tree/v1.0.3

MMA8453_n0m1: the `sentence` field must be in one long line (it should not span over multiple lines as in https://github.com/n0m1/MMA8453_n0m1/blob/master/library.properties#L6), please fix it and tag it again (removing the old tag is fine too)

Comp6DOF_n0m1: same problem here: https://github.com/n0m1/Comp6DOF_n0m1/blob/master/library.properties#L6

SST25VF: ok, added

Sleep_n0m1: ok, added

NicoHood, 2015-07-15:
May I set a reminder for my requests here?

https://github.com/arduino/Arduino/issues/3342
https://github.com/arduino/Arduino/issues/3230

nullboundary, 2015-07-16:
Ok, those 3 should be good now. Made the fixes you suggested. 

You guys really need to automate this submission process. Must be a lot of work to do this all manually.

cmaglie, 2015-07-16:
The libraries are online now (well they were already from about 10 hours ago when you submitted the change).

> You guys really need to automate this submission process.

We have it in the todo list, BTW the amount of requests seems to be stabilizing after the initial peak, at the current rate can still be easily handled manually.

Thanks!

closed 2015-07-16 by cmaglie