arduino/Arduino #3531: SoftwareSerial sketch makes some characters dissappear.

Avamander, 2015-07-15:
The default SoftwareSerial sketch. With a few modifications, just makes certain characters dissappear that causes connected modem to dial wrong numbers. 

For example (carriage return enabled, but it does not affect the result) "ATD54321054;" results in "ATD " printed back to the computer, "ATD543210540000000000;" results in "ATD0000". 

What is going on?

```
/*
  Software serial multple serial test

 Receives from the hardware serial, sends to software serial.
 Receives from software serial, sends to hardware serial.

 The circuit:
 * RX is digital pin 10 (connect to TX of other device)
 * TX is digital pin 11 (connect to RX of other device)

 Note:
 Not all pins on the Mega and Mega 2560 support change interrupts,
 so only the following can be used for RX:
 10, 11, 12, 13, 50, 51, 52, 53, 62, 63, 64, 65, 66, 67, 68, 69

 Not all pins on the Leonardo support change interrupts,
 so only the following can be used for RX:
 8, 9, 10, 11, 14 (MISO), 15 (SCK), 16 (MOSI).

 created back in the mists of time
 modified 25 May 2012
 by Tom Igoe
 based on Mikal Hart's example

 This example code is in the public domain.

 */
int val;
int a = 10;
#include <SoftwareSerial.h>

SoftwareSerial mySerial(A1, A2); // RX, TX

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  pinMode(A1, INPUT);
  pinMode(A2, OUTPUT);

  Serial.println("Goodnight moon!");

  // set the data rate for the SoftwareSerial port
  mySerial.begin(9600);
}

void loop() // run over and over
{
  if (mySerial.available()) {
    Serial.write(mySerial.read());
  }

  if (Serial.available()) {
    int val = Serial.read();
    Serial.write((char)val);
    mySerial.write(val);
  }
}
```

NicoHood, 2015-07-15:
It would be nice to know what IDE version you are using since @matthijskooijman provided some fixes with 1.6.1 or 1.6.2

Avamander, 2015-07-15:
1.6.6 12/07(not sure about the date)

mikaelpatel, 2015-07-15:
@Avamander 
Have you tried adjusting the baudrates in your example sketch? What happens if both are for instance 9600?

facchinm, 2015-07-16:
Hi @Avamander , I've tested your sketch and it behaves as you report only if you send bursts of data; using `minicom` or another terminal emulator (that sends characters as soon as you type them) all chars are received correctly.
If you need to send and receive burst instead, buffering is the way to follow. Your sketch could simply become 

```arduino
#include <SoftwareSerial.h>

SoftwareSerial mySerial(A1, A2); // RX, TX

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  pinMode(A1, INPUT);
  pinMode(A2, OUTPUT);

  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  Serial.println("Goodnight moon!");

  // set the data rate for the SoftwareSerial port
  mySerial.begin(9600);
}

char val;
char rx_buf[128];
char tx_buf[128];
int i_rx=0;
int i_tx=0;

void loop() // run over and over
{
  if (mySerial.available()) {
    val = mySerial.read();
    rx_buf[i_rx++] = val;
    Serial.print(i_rx);
    if (val == '\n' || val == '\r') {
        Serial.write(rx_buf, i_rx);
        i_rx = 0;
      }
  }

  if (Serial.available()) {
    val = Serial.read();
    tx_buf[i_tx++] = val;
    if (val == '\n' || val == '\r') {
        Serial.write(tx_buf, i_tx);
        mySerial.write(tx_buf, i_tx);
        i_tx = 0;
    }
  }
}
```

where `\n` and `\r` are the line terminators.

Avamander, 2015-07-16:
I used IDE's built in terminal.

facchinm, 2015-07-16:
The builtin terminal sends all the characters at once when you press `ENTER`, so you are falling into the "burst" case. Could you try my sketch and report if it works for you?

Avamander, 2015-07-21:
@facchinm 
Sorry for not responding earlier. The new sketch you provided does indeed work. I suggest you'd update the Arduino IDE's builtin sketch too. Also, does the Arduino IDE's serial monitor not have delay between sent characters?

closed 2015-07-21 by Avamander

matthijskooijman, 2015-07-23:
The problem is really the baud rates: SoftwareSerial runs at a lower baudrate than the hardware serial. Since SS blocks the CPU completely (interrupts disabled) while sending a byte, and because the SS baudrate is a lot lower than the hardware baudrate, this means that in a single SS byte time, multiple bytes are received on the hardware port. Since interrupts are disabled and there is only a single byte hardware buffer, a lot of bytes get overwritten.

The solution proposed by @facchinm works, but only if the data sent is a single line, followed by a pause (allowing SS to write out the buffer before new things are received).

Overall, the problem is just that SoftwareSerial takes up the CPU and disable interrupts and it depends greatly on the situation whether that will cause problems or not, or what fix is appropriate. I don't think there is a single magic solution here, though perhaps the comments and documentation can be improved.