arduino/Arduino #3532: Commented code highlighted wrong

NicoHood, 2015-07-15:
If you put a single, capital character at the start of a /**/ comment the underlaying paragraph is not grey anymore. Not all letters are affected, only a few, and only capital once.

Example:
```
/*
Test
A
Test
*/
```

per1234, 2015-07-16:
That example doesn't work for me, but I think I know why. I have noticed this happens when an unindented keyword is in a multi-line comment. I'm guessing Test is in one of your keywords.txt files and not mine. For example:
```
/*
this is gray
 void
this is still gray
void
this is not gray
*/
```
[image]

NicoHood, 2015-07-16:
Not talking about the "test", rather the "A" and other letters. But yes, that is the issue. This shouldnt happen.

per1234, 2015-07-16:
Is "A" a keyword in your installation?

NicoHood, 2015-07-16:
Yes. And other capital letters. Dont know why. Might be because of some libraries, but I will not search them now.

lcafaro, 2015-07-19:
This should help you find the culprit:
```find . -type f -name keywords.txt | xargs grep '^A '```

Anyway, I think this should be marked as not a bug.

[image]

carlosperate, 2015-07-19:
If the code highlighting fails using default keywords as shown by per1234, then it is a bug.

lcafaro, 2015-07-19:
You're right, I missed that bit. Apparently this only happens with that style of commenting.
Looks like it's related to our implementation of ```rsyntaxtextarea.TokenMap``` (e.g. PdeKeywords.java) since I can't reproduce on a plain Java implementation of rsyntaxtextarea.

ffissore, 2015-09-17:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 94d1829b (2015-09-17): If a token is of some COMMENT_* type, don't use keywords. Fixes #3532
closed 2015-09-17 by ffissore