arduino/Arduino #3534: keywords.txt

MarkusLange, 2015-07-15:
INPUT_PULLDOWN LITERAL2 Constants RESERVED_WORD_2

is missing INPUT_PULLDOWN has no color

facchinm, 2015-07-16:
Hi @MarkusLange  , indeed `INPUT_PULLDOWN` is not a valid keyword as it's not declared anywhere in the IDE :smile: 
Where did you find it?

Chris--A, 2015-07-16:
A custom library maybe?

facchinm, 2015-07-16:
I also thought it was in a custom library, but if the code is declared inside a custom core you can't specify a custom keyboard.txt (issue #755). Leaving open for reference

MarkusLange, 2015-07-16:
Hi Chris, hi facchinm,

I found in ...\samd\cores\arduino\wiring_digital.c from the new ZERO Board, the ZERO has of pull-up and pull-down resistors on the pins.

facchinm, 2015-07-16:
Well, this is a bit awkward :grin: I didn't grep inside Zero core actually. 
We can't add the keyword platform-wide (if we don't provide an alias of `INPUT` as `INPUT_PULLDOWN` of course) but I'll push to solve #755 and then we'll add the keyword only to Zero core.
Thank you very much Markus!

MarkusLange, 2015-07-16:
Not for that. I tripped over it because of a similar issue (https://forum.pjrc.com/threads/29140-Bug-in-teensy-installer-for-arduino-1-6-5-r2) and my messing around on the Onewire lib to get the Zero on duty, works so far, (http://forum.arduino.cc/index.php?topic=331220.msg2297555#msg2297555) but I don't get the basic operations, direct register access working.

PaulStoffregen, 2015-07-16:
I have long wanted this #755 feature for Teensy.  Of course, not enough to dig into the IDE source to implement it....

Until recently, there's been relatively little activity in the global keywords.txt file, so I've just installed a patched copy.  Obviously that's not so good.

Ideally, the keywords.txt from core libraries could apply in addition to the global one, much like how the keyboard.txt from libraries do.  All that's needed is more hours in the day for someone to actually implement this in the IDE.....

closed 2017-07-04 by facchinm