arduino/Arduino #3537: Remove StringToIntRGB Example

00alis, 2015-07-16:
Hi! 
Simone Maiocchi is reviewing the whole bulk of the Built-in Examples (we are updating that whole section of the website) and he suggested we remove the StringToIntRGB example since it almost identical to ReadASCIIString and maybe it was made obsolete by it. 

What do you think?
@ffissore @cmaglie

cmaglie, 2015-07-16:
The two examples are:

https://github.com/arduino/Arduino/blob/master/build/shared/examples/04.Communication/ReadASCIIString/ReadASCIIString.ino

and

https://github.com/arduino/Arduino/blob/master/build/shared/examples/08.Strings/StringToIntRGB/StringToIntRGB.ino

Both sketch do the same thing (read a string and light up an LED with the requested color) but:
- ReadASCIIString do it in a more "naif" way, better for beginners to understand code, but prone to timings errors on production projects
- the StringToIntRGB example instead is a complete project: it does the parsing in a more sophisticated way and contains also a [Processing's sketch](https://github.com/arduino/Arduino/blob/master/build/shared/examples/08.Strings/StringToIntRGB/StringToIntRGB.ino#L110) that `draws a color wheel and send a serial string with the color you click on` 

IMHO both the examples have their purposes, maybe StringToIntRGB should be renamed and moved away from the "String examples" folder so it's clear that is a complete project?

00alis, 2015-07-16:
Thank you so much! Going to pass this info along to Simone and see if he is on Github too :)

00alis, 2015-07-16:
here he is @SimonePDA

SimonePDA, 2015-07-16:
Hi Cristian,

I agree if we can move the more complex example somewhere else. I find the examples that involve other programming environments a little distracting for the normal user because they open up more questions than answers.

Maybe it is time to have a "Arduino and Processing" section that collates all the examples that interact with Processing not only with the Firmata stuff.

simone

SimonePDA, 2015-08-27:
Hi again,
Could you please detach the StringToIntRGB from the Strings examples so that we put it somewhere else (we will find a place later on)?

I'm finalizing the Tutorials and that example is going to be orphaned and left out from the examples list

simone

agdl, 2015-09-03:
@SimonePDA @cmaglie @00alis I'll remove it and make a PR now!

closed 2015-09-03 by agdl