arduino/Arduino #3539: [Idea] New autobackup function

NicoHood, 2015-07-16:
Regarding to this issue I had an idea:
https://github.com/arduino/Arduino/issues/3174

The ide has an autosave on compile function. This overwrites the file.
I am one of those guys who open old sketches, ruin them to try things out and close them without saving. So I do not want to enable this option for me.

A backup however would be nice if the IDE crashes again.
I'd go for a ~sketch.ino file like most editors under linux do. Then you always have a backup if the ide crashes, but its not saved directly. The backup can be saved in the directory itself or somewhere else. On exit it should clean those files. If it crashes, the next time it can ask if it should recreate those files or compare etc.

VogonJeltz, 2015-07-16:
I double that!
Please always save to temp files for compiling, no overwriting of the current file!

NicoHood, 2015-07-16:
That is exactly what I wanted.

Chris--A, 2015-07-16:
Yes, this is something that is bugging me too. Definitely a needs an option in the preferences, or command line switch at the minimum. I can put up with the 'really' annoying save dialog that pops up just to test a single line of code... however loosing a project because of testing or inadvertent copying to the wrong window is going to be sad day.

pfeerick, 2015-07-24:
Would love to see this also... having the IDE save over the top of something just because I'm trying out a few different lines is getting pretty annoying... it should save to a temporary file, compile that, and if I try to close the sketch without saving the file, prompt me. Now that I have come to grips with how it works, it is annoying, but I lost quite a bit of code when fiddling with different sketches because of this behaviour. I think that would also neatly incorporate the backup feature that @Nicohood is requesting.

DKWatson, 2017-07-13:
It's been 2 years since there's been activity here. Don't know if that's good news or bad. In any event, the issue still exists. I've just lost several hours work (not for the first time) because the IDE overwrites existing files without asking. There seems little sense in trying new things at the expense of old. Yes I should be more diligent in backing up my own work and usually I have four to six copies scattered across a few machines, but now and then you get carried away. The coffee is hot, the music is good, the ideas keep flowing and then all of a sudden you try a validate and everything goes south, and there's no backup because the IDE just erased it. It wouldn't perhaps be so bad if the undo worked and one could erase a host of changes, but the undo feature of the IDE is a catastrophic nightmare.

At least give us the option of saving the previous copy, please.

per1234, 2017-07-13:
@DKWatson if you have discovered any bugs in the Arduino IDE's Undo using the latest version of the Arduino IDE then please do report them in a separate issue, after first searching to be sure it hasn't already been reported. Thanks!

One thing to be aware of is that you can turn off the (IMO unpleasantly astonishing as a default) feature of the Arduino IDE automatically saving before compilation at **File > Preferences > Save when verifying or uploading**.

I strongly recommend that everyone start to use version control software such as Git as soon as they get past the beginner stage. The time to learn and implement proper version control and an automated backup system will surely pay off many times over in the long run. That's not to say I'm necessarily against this feature request.

DKWatson, 2017-07-13:
Hey, cool, a response straight away, and you even sound like you're human, not machine generated. I come from a heavy Autodesk environment where every time you recompile or re-render or save or anything that changes the file you started your session with, there's an autosave with an incremental number tacked on the end of the filename. Now what gets annoying, is having to cleans house in your directory as multi-megabyte files consume disk space rather rapidly. Th point is however, that if I choose, I can go back to anywhere along my timeline and recover my work from any stage. As .ino files tend to be only 10kB or so, it's not that big an issue. It would be nice if that option existed.

flanwan, 2020-11-27:
Hello

I'd like to support this request too.

I store of course minor versions and also patches on functional changes. But when I am at formating the LCD outputs spread over my code, then this is a constant build-and-test-circle where I build every minute a new version. Unfortunately it easily happens that some find&replace or wrong copy&paste spoils my code (I think everyone here once started searching the missing bracket 400 lines of code above...). Working on text output format is nothing which anyone would cover by git-versioning.

So a simple jump back to recent build versions would be really appreciated. 
My suggestion is quite simple. When you store the sketch to the *ino-file, then also store it to a file that has a timestamp in the file name. Something like
mysketch.ino
2020-09-28_22:31:04_mysketch.ino
2020-09-28_21:58:24_mysketch.ino
...
This avoids any numbering and also describes files better than simple numbers (I remember, what I did 15 minutes ago, but I don't remember what I did in backupfile number 123)

Best regards
Florian Anwander

pfeerick, 2020-11-30:
It was partly failure to make changes as basic as this that ultimately have given up in the Arduino IDE, and instead use VSCode + PlatformIO. VSCode is a much more mature and featured IDE, allowing multi-undos even after saving, and even preserves unsaved changes (or perhaps just files?) across restarts of the IDE. 

With the Arduino Pro IDE being based on Eclipse Theia, a lot more features should be come standard, and the rich extensions platform should make it so you can make your IDE do basically whatever you need it to.