arduino/Arduino #3543: Add a scrolling bar to the tab menu

clefranc, 2015-07-17:
@ffissore This is a follow up of the issue https://github.com/arduino/Arduino/issues/2765 AKA "**How do you know which tab you're editing?**" that was resolved with https://github.com/arduino/Arduino/commit/2339049c0fd0cd7d215adc659f044a999a9dfeca, thank you again.

Working with the IDE is now (in my case) day and night. But there is a last one improvement I wish to see in the IDE: a scrolling bar in the tab menu. I think the top and bottom arrow in the tab menu should be replaced by a scroll bar. Not only it's more intuitive, it is ***WAY SPEEDIER***.

Also, the tab bar should have a left and right arrow to effectively scroll the tabs (like asked in https://github.com/arduino/Arduino/issues/1193, it was in 2012).

Is this feasible?

ffissore, 2015-08-04:
Feasible, yes. See http://www.wideskills.com/java-tutorial/java-scrollable-popup-menu
When will this be done? Hard to say ATM. But I hate those arrows too. I'll try to push this up in the list

clefranc, 2015-10-05:
@ffissore Any development on this feature?

lmihalkovic, 2015-11-21:
@clefranc  could [this](#4083) be a valid alternative?

clefranc, 2015-11-21:
@lmihalkovic Sorry, your link point here... What alternative?

matthijskooijman, 2015-12-09:
I think he meant #4083

clefranc, 2016-11-10:
@ffissore Is it still on the to-do list?

ffissore, 2016-11-10:
@clefranc sorry you missed the news. I no longer work for arduino. See https://groups.google.com/a/arduino.cc/forum/#!msg/developers/YJLX6AZHem4/kpsLef7XBAAJ

sbrl, 2017-03-01:
Some support for larger projects with more files would be great. What's the status of this? And why was it closed?

per1234, 2017-07-18:
I'm reopening this because I think it was closed due to a misunderstanding. Just because ffissore isn't working on it doesn't necessarily mean that Arduino has declared this won't happen. It should stay open until it's done or officially declared wontfix.

The tab menu still uses the arrows at top and bottom for scrolling rather than a scrollbar in Arduino IDE 1.8.3.