arduino/Arduino #3544: [Feature Request] Specify path to preferences.txt and sketchbook folder in command line

liudr, 2015-07-18:
Here is what I mean:

When you run Arduino.exe in command line or GUI, add the feature so I can do command-line options:

"Arduino.exe pre=c:\blah\preferences.txt sketchbook=c:\blah\sketchbook1"

This makes Arduino extract preferences from a specific location if this option is used, overwriting the default location in AppData\Roaming.

The second option makes Arduino use the specified folder to search for sketchbooks. This option should override the path found in preferences.txt

Reasons:
1. As we all know, the entire Arduino IDE's custom features are in the sketchbooks folder, contributed libraries, 3rd-party hardware, and all the sketches. If you can specify a different sketchbook folder, you get a whole different experience. I teach and do workshops with arduino boards and my own compatible boards. When I teach or present, I'd like to show the least amount of sketchbooks and be able to control what sketchbooks are shown and what hardware and libraries to load. Best way to achieve this is to have a different sketchbook folder just for a particular class and workshop. Having an Arduino.exxe shortcut that has the parameters for each class/workshop will be very nice. It separates my own work, and other classes from those that don't need to see.

2. In different projects, you may find yourself using different versions of forks of the same library, which are not compatible. If you can specify which sketchbooks folder to load, in a shortcut, you just call the shortcut "Arduino with datalogger project settings" and it loads the correct libraries. The alternative is to constantly changing the folder option and restart arduino. I'll give you an example: adafruit sells DS1307-based real-time clock modules and data logging shields. It runs RTClib library. It also sells DS3231-based "chronodot" modules that runs a fork of a fork of the RTClib. They are called the same name and can't coexist in one libraries folder. I have projects that need DS1307 and others that need DS3231. Having a command-line option will be wonderful way to separate them without hassle.

matthijskooijman, 2015-07-23:
What you're requesitn is already possible. See the manpage here: https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

In particular, see the `--preferences-file /path/to/file` and `--pref sketchbook.path=/path/to/dir` options. I'm closing this issue, feel free to reopen or comment if I misunderstood and you need something else or more.

closed 2015-07-23 by matthijskooijman

liudr, 2015-07-29:
Matthijskooijman,

That was a start. The potion you gave me permanently set the pref name's value. Arduino IDE only flashes its banner briefly without opening a window. That's not what I wanted. What I wanted is to OPEN Arduino IDE with a set of preference values that I can specify in command line.

Do you use windows? In windows, you can create shortcuts that specify program to run and command line options for the program. This way depending on which shortcut you click, the program may run with different options turned on or off. What I want to turn on or specify is to run the GUI with a specific sketchbook path. Say for workshop A, introduction to arduino to high school kids, I want to have sketchbook folder that contains some robotics, some fun LED stuff etc. Then for some teacher workshop B, I want to show them data logging with photogates. If I can specify sketchbook location, I don't have to keep modifying arduino's preferences all the time.

matthijskooijman, 2015-07-30:
Ah, I assumed you wanted to a one-off compile with a custom sketchbook path. If you want to open the IDE with a custom pref, this should be possible, but it seems it is currently broken (reported at #3511). Seems you'll need that bug fixed to make it work like you want.

As for the preference being permanent, there is currently no way to open up the IDE and only set a preference for that invocation. This is only possible for one-off compilations / uploads, because when the IDE GUI is closed, the preferences are always saved. If using --pref would prevent this saving from happening, things would get confusing (preferences made through the preferences dialog would also no longer be saved. The code could track what preferences to (not) save, but that would get complicated quickly and simply hasn't been implemented yet.

liudr, 2015-07-30:
Great! At least I have something to look forward to when the bug is fixed. The man page does say the IDE will NOT open whenever any option is given though. Right, permanent is ok. If the feature works, I don't worry about permanently changing the pref back and forth with different shortcuts.