arduino/Arduino #3545: unable to recognize the port

Manisha-Manimaran, 2015-07-18:
after making a connection with intel galileo2 boarr ..my lap is not showing port information....

Avamander, 2015-07-21:
Have you tried the latest hourly build?

Manisha-Manimaran, 2015-07-24:
.I have tried but,I'm not getting the port on the tools bar.is there any other possiblities ??

Avamander, 2015-07-24:
Have you installed all the software the installation manual told you?

Manisha-Manimaran, 2015-07-26:
yes

facchinm, 2015-07-27:
Have you tried installing the specific Galileo drivers? You can find them [here] (https://downloadcenter.intel.com/downloads/eula/24748/Intel-Galileo-Firmware-and-Drivers-1-0-4?httpDown=http%3A%2F%2Fdownloadmirror.intel.com%2F24748%2Feng%2FIntelGalileoWindowsSerialDrivers.zip)

ffissore, 2015-08-03:
@Manisha-Manimaran any news?

closed 2015-08-05 by ffissore