arduino/Arduino #3547: new packages on 1.6.5 are broken

xxxajk, 2015-07-19:
1.9.2 and 1.9.3 avr board packages are broken.
There is no sketch dir created during compile on IDE 1.6.5.
```
/opt/arduino-1.6.5/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld: cannot open output file /tmp/build7121033705008165262.tmp/sketch/sketch_jul17a.cpp.elf: No such file or directory
collect2: error: ld returned 1 exit status
Error compiling.
```
My guess here that this is going to be supported in 1.6.6?

matthijskooijman, 2015-07-23:
Right, the paths for the sketch changed, but the old IDE doesn't contain the right mkdir yet. IIRC there was an issue to track unexpected problems as a result of this path change, but I can't find it right now. IIRC @ffissore was working on it.

cmaglie, 2017-03-30:
I'm closing this one as wontfix, since those packages are now very very old.

closed 2017-03-30 by cmaglie