arduino/Arduino #3548: linux 32bit avrdude should be static

xxxajk, 2015-07-19:
/opt/Arduino/hardware/tools/avr/bin/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory

avrdude should really be compiled static in order to avoid this sort of nonsense.

facchinm, 2015-07-20:
Hi @xxxajk, did you update `libncurses` recently? However you are right, the bundled 32bit package only contains `libusb` library binaries. Would you mind moving the issue from here to https://github.com/arduino/toolchain-avr/issues

xxxajk, 2015-07-20:
A symlink to my current library works. Point is that you should not have to do this. 
If you link statically, you ensure that the program will work whether-or-not someone has any particular library installed, at the cost of a slightly less larger distributed binary.

xxxajk, 2015-07-20:
Oh yeah, and I do not mind moving the issue.

closed 2015-07-21 by facchinm