arduino/Arduino #3550: Arduino Due hardware folder misplaced on Windows 7

KepecsLab, 2015-07-19:
In the version 1.6.3 notes, "AVR bundle files (core files and toolchain) is again inside IDE "hardware" folder, instead of unpacked into user's home folder".
This creates confusion, because the "SAM bundle files" are still in /Users/Username/AppData/Roaming/Arduino15/packages/arduino/hardware/

This was noted in the forums:

Chris--A, 2015-07-19:
There has been a discussion about a new location for all I think: https://groups.google.com/a/arduino.cc/forum/#!topic/developers/QigfElnAMJ4

Might answer some questions for you.

facchinm, 2015-07-20:
Starting from 1.6.5 the default location of downloaded cores has moved from `Appdata\Roaming` to `%LOCALAPPDATA%\Arduino15` to avoid breaking Microsoft specifications (#3404). 

Only the AVR core (which is still bundled with the IDE) should be inside `hardware` folder; every other core will be located in `%LOCALAPPDATA%\Arduino15` if downloaded from the Board Manager or in `sketckbook/hardware` folder if extracted manually.

closed 2015-07-20 by facchinm